RasPlex/RasPlex #413: CEC stops working after tv is reset

rbeall, 2015-08-15:
CEC stops working after my samsung tv is reset.  I recently upgraded to 0.7.0. but prior to that, everything CEC related as worked fine.  With this upgrade, CEC stops working when I reboot my tv (with the raspberry pi staying on).  Is this a setting somewhere that got changed with the build or is it a bug?

Thanks

rbeall, 2015-08-15:
I took a look at the CEC settings per some other troubleshooting forum advice and noticed some more data points.  The settings are set to ignore the tv shutdown and I think it is.  Because when I turned the tv back on CEC started working about 5 seconds later (like it normally would) but it only worked for like 3 button pushes and then stopped working.  If I reboot the pi it works fine the whole time.

rbeall, 2015-08-15:
I booted into 0.6.0 and CEC is working fine in all scenarios (even after rebooting tv).  I copied down the CEC settings and used the same settings in 0.7.0 and it is bugging out after 2-3 button pushes after the tv has been reset.

Snekatis, 2015-08-18:
I have the same problem.

mihalski, 2015-09-01:
I don't think it's a CEC problem if it's anything like what I am encountering. Similar symptoms but it takes longer for the TV to ignore the remote (regardless of whether it's on or off).

I have noticed that if I connect with my iPhone Plex app as a remote (without even having to press any buttons after connecting) the CEC controls come back also. So the problem seems to be more generally remote related rather than CEC specific. At least that's my theory.

rbeall, 2015-09-01:
All I know is I downgraded my firmware and my hardware works just fine.  Your iPhone app solution might be a clue that some part of the code is going to sleep.

JHurk, 2015-09-02:
Same problem here

rbeall, 2015-10-26:
I just loaded the latest 7.1 on a new rasppi2 on brand new hardware to see if anything has been updated or if the new hardware would get different results.  I'm still getting the lock up after the tv shuts off.  When the tv boots back up, CEC works for like two button pushes and then stops working just like before.

Potential work-around.  I used the plex app on my iPhone to do some more video streaming testing etc and the iPhone remote (as expected) worked fine.  However, after watching a tv espisode, the cec was working fine afterwards and also worked after tv was shut off the second time.  It was like using the iPhone app got it 'unstuck'  ??  I don't have much proof that this work around works consistently but it has worked for two shutdowns now.  I'll post any other thoughts if I find anything.

NedtheNerd, 2015-11-09:
Can you check in latest release -1.0.0 - and advise.

NedtheNerd, 2015-11-18:
No further feedback, closing issue.

closed 2015-11-18 by NedtheNerd