RasPlex/RasPlex #414: Audio stutter Rasplex 0.7.0 on raspberry p+

dsfser, 2015-08-15:
Hello,

I would like to let you know that version 0.7.0 stutters when playing a simple mp3.

I have also installed 0.6.0 and when i play the same mp3 it works perfectly without stuttering.

Also i have done some research myself:

i have read that 0.7.0 uses PAPlayer:  ( PAPlayer is now used for music )

So i created the file and put this in it:
```
<advancedsettings>
     <audio>
           <defaultplayer>dvdplayer</defaultplayer> 
     </audio>
</advancedsettings>
```

In the advancedsettings.xml file located: ~/.plexht/userdata/

The music still stutters but alot less than with paplayer.

My pi:
Raspberry pi B+

anyone ideas?

Thanx

dsfser, 2015-08-15:
oke so i did some more research and found out that when i enable the option:
Solby Digital Plus (E-AC3) capable receiver

all the stuttering is gone on version 0.7.0

strange.

dsfser, 2015-08-15:
also i added the following to my advancedsettings.xml:

<advancedsettings>
        <audio>
                <defaultplayer>dvdplayer</defaultplayer>
        </audio>
        <network>
                <buffermode>1</buffermode>
                <cachemembuffersize>52428800</cachemembuffersize>
        </network>
</advancedsettings>

this did fix the problem with the stuttering.

closed 2015-08-15 by dsfser

LegeDoos, 2015-10-01:
I have the same stuttering problems in my Rasplex with Raspberry pi. Will this solution you mention be in a next release?

dsfser, 2015-10-02:
I dont think this is a good sollution to put in a new release. They need to fix it using a other sulition. Strange thing is that kodi (openelec) doenst have any problems playing these files on the same raspberry.