RasPlex/RasPlex #415: No cover art

StuartRob, 2015-08-19:
No matter what I do I get no cover art. Web plex and on other devices I can see cover art for my media library. Plex is up to date and I've rescanned all libraries. I have a new install of rasplex 0.7.0, not an update: Raspberry pi 2, 4gb sd card. Partial debug log at http://pastebin.com/8dx9xRc6

UnpaidMoneyMike, 2015-08-31:
I am having the exact same issue as well. Have you found a fix?

Kwiboo, 2015-09-01:
@StuartRob: Please enable debug logging in Preferences -> System -> Advanced, your current log do not have enough details to pinpoint the source of your problem.

StuartRob, 2015-09-05:
Ok, I have a whole new install of plex media server (0.9.12.11 on Freebsd 10.2) with new cover art and a whole new install of rasplex 0.7.0 on rpi 2. Playback works as expected. I can see cover art online and on various devices but not rasplex.

I've enabled debug logging, partial log at http://pastebin.com/GVJbZsgT. I have the whole log but it's 2.5mb, too big for pastebin.

Kwiboo, 2015-09-05:
@StuartRob: Thanks for the log, it appears that your RasPlex device and PMS is located on the same subnet but for some reason is using your external ip+port to communicate with PMS.

Your RasPlex device is currently not picking up any network discovery response from your PMS.
When RasPlex gets the a response from the network discovery it will switch to and use the local connection to your PMS.

Please test the following and report back.
1. Remove `/storage/.plexht/userdata/Database/PlexServerCache4.db` and restart RasPlex.
2. Ensure the `Enable local network discovery (GDM)` setting is enabled on your PMS and that incoming multicast to `239.0.0.250:32413` is not blocked by any firewall on your PMS.

StuartRob, 2015-09-08:
Okay, I removed the PlexServerCache4.db and checked my ports and firewall and rebooted. It had no obvious effect; still no cover art. I checked internet connections in the jail in which my plex server runs and everything seemed fine (I have ifconfig and resolve.config and nsookup info if that's helpful).

For me cover art has been a problem since 0.6.0 and the introduction of managed users. In 0.5.1 I had one user and art work was fetched. In 0.6.0 and later I had two users and no art. I have removed the second user, chosen automatic sign in and rebooted and now I have cover art for all my media. 

Later, when I'm at home, I'll upload my new debug log to git.

NedtheNerd, 2015-09-08:
Have you checked the file permissions and ownership for the plexmediaserver directory on your server?

NedtheNerd, 2015-09-08:
Also, have you searched the FreeBSD forum http://forums.plex.tv/categories/freebsd as this is more likely related to your PMS install and we know it's not a Rasplex issue.

StuartRob, 2015-09-08:
I've looked at lots of possible solutions on various forums but the change that I made and that made a difference was having only one Plex user. 

I have Plex 0.9.12.11 in a Freebsd jail running Freebsd 9.3. Both the resolve.conf in the jail and outside  point to my router as a name server. I can connect to the internet fíne and can access my plex server remotely. I haven't forwarded any ports. Below are responses to ifconfig on the host and in the jail:

```ifconfig on host

re0: flags=8843<UP,BROADCAST,RUNNING,SIMPLEX,MULTICAST> metric 0 mtu 1500
	options=8209b<RXCSUM,TXCSUM,VLAN_MTU,VLAN_HWTAGGING,VLAN_HWCSUM,WOL_MAGIC,LINKSTATE>
	ether 20:cf:30:43:da:35
	inet 192.168.0.16 netmask 0xffffff00 broadcast 192.168.0.255 
	inet 192.168.0.91 netmask 0xffffff00 broadcast 192.168.0.255 
	inet 192.168.0.95 netmask 0xffffff00 broadcast 192.168.0.255 
	nd6 options=29<PERFORMNUD,IFDISABLED,AUTO_LINKLOCAL>
	media: Ethernet autoselect (1000baseT <full-duplex>)
	status: active

ifconfig in jail

re0: flags=8843<UP,BROADCAST,RUNNING,SIMPLEX,MULTICAST> metric 0 mtu 1500
	options=8209b<RXCSUM,TXCSUM,VLAN_MTU,VLAN_HWTAGGING,VLAN_HWCSUM,WOL_MAGIC,LINKSTATE>
	ether 20:cf:30:43:da:35
	inet 192.168.0.95 netmask 0xffffff00 broadcast 192.168.0.255
	media: Ethernet autoselect (1000baseT <full-duplex>)
	status: active
lo0: flags=8049<UP,LOOPBACK,RUNNING,MULTICAST> metric 0 mtu 16384
	options=600003<RXCSUM,TXCSUM,RXCSUM_IPV6,TXCSUM_IPV6>
```

I don't understand networking or jails enough to tell whether the information here shows that this is a freebsd and/or a jail problem. I've linked my new plex home theater log above, perhaps this helps clarify where the problem lies.

NedtheNerd, 2015-11-10:
Can you check this issue with the latest release - 1.0.0 - and advise.

NedtheNerd, 2015-11-18:
No further feedback, closing issue.

closed 2015-11-18 by NedtheNerd