RasPlex/RasPlex #416: Plug and Play Keyboard Support, Part 2

m00dawg, 2015-08-19:
I noticed in issue 384 you mentioned plug and play keyboard support will never work? Curious as to why? OSMC doesn't have issues with that? Or, more to the point, I can use a KVM to switch to/from my rPi and my desktop machines via OSMC. I cannot do this in RasPlex which makes it very difficult to use with my specific setup.

Kwiboo, 2015-09-01:
This should be fixed in next release.

m00dawg, 2015-09-01:
:rocket:

NedtheNerd, 2015-09-08:
@m00dawg 

Can you check in latest release 0.7.1 and advise.

m00dawg, 2015-09-09:
Works like a champ!

There is a delay when switching to the rPi before the keyboard works (about 5 seconds), which isn't a big deal. Other than that, it seems to work without issue!

Thanks!

NedtheNerd, 2015-09-09:
Delay is normal. Closing issue.

closed 2015-09-09 by NedtheNerd