RasPlex/RasPlex #418: Filter/Sort with Multiple Plex Servers

blix88, 2015-08-20:
**[Description]**
I have two plex servers named PX-01 and PX-02. When I navigate to the Movie library on PX-01 and apply a filter/sort it applies successfully. I then go to the Movie library on PX-02 and successfully apply a filter/sort. However when I go back into the Movie library on PX-01 to apply a filter/sort, the filter/sort screen is still applying to the PX-02 server. Any filters or sorts are not applied to the PX-01 server, but instead applied to the PX-02 server. 

**[Screenshot]**
[image]

**[RasPlex Info]**
    Model: Raspberry Pi 2 (RPi2) Quad-Core 900 MHz 1GB RAM
    Network Connection: WiFi
    Version: 0.7.0
    Log File: http://pastebin.com/409PVCbv
    Install Method: Linux dd
    
**[PlexServer Info]**
    Name: PX-01
    Version: 0.9.12.9
    Server Location: Local to RasPlex

**[PlexServer Info]**
    Name: PX-02
    Version: 0.9.12.9
    Server Location: Local to RasPlex

NedtheNerd, 2015-08-20:
Known issue with filters, inherited from PHT 1.4.1, fix coming in next release.

NedtheNerd, 2015-09-08:
@blix88 

Can you check in latest release 0.7.1 and advise if fixed.

NedtheNerd, 2015-09-11:
Filters fixed in 0.7.1, closing issue.

closed 2015-09-11 by NedtheNerd