RasPlex/RasPlex #419: Crash upon login

will32, 2015-08-20:
I tried installing 0.7.0 under BerryBoot. I came across a problem, so tried installing 0.7.0 directly to an SD card, but I get the same problem.

It boots OK, I go through the set up and then Home Theater has only two menu options - Search and Switch User. When I choose Switch User, it shows my users within my Plex Home. I choose one and after about a second the screen goes blank, my TV loses signal, and a couple of seconds later I see the "RasPlex" splash screen again. So I'm guessing that the OS doesn't crash, but just Home Theater? I've tried fiddling with boot options and the same happens every time.

Log file: http://pastebin.com/1HiPJ9rX

Crash ID: 1741197
Size of Raspberry Pi: 1GB (Pi 2 Model B) 
Type of SD card: SDXC Class 10 64GB
Internet connection type: Ethernet
Version of RasPlex: 0.7.0
Log files (ensure that you enable "debug logging in Preferences -> System -> advanced" and try to reproduce the issue)
How did you install rasplex: win32 disk imager
Version of Plex Server: 0.9.12.9
Type of plex server: Local
Power supply type and power rating: Non-switched 2A @ 5V
Comment on system load: Load average 0.40 0.32 0.20. At time of crash, about 5% CPU, 480MB free RAM.

Kwiboo, 2015-09-08:
@will32 Please test with RasPlex 0.7.1 also enable debug logging in Preferences -> System -> Advanced.

will32, 2015-09-17:
Fixed for me in 0.7.1.

closed 2015-09-17 by will32