RasPlex/RasPlex #420: How to setup a build environment? Page missing

sjorge, 2015-08-28:
Hey All,

I was looking into seeing if I could reuse the rasplex stuff to spit out a more up to date OpenELEC+PlexHT for x86(_64). However it seems the link on setting up the build env (going to the openelec folks) is now dead.

Regards

NedtheNerd, 2015-08-28:
The last stable code here is for Rasplex 0.4.1 which is Plex HT 1.2.2, so not really going to provide you with anything more that what's already there in the latest OpenELEC + PHT for x64. Have you raised this on the forums http://forums.plex.tv/discussion/70330/openelec-presentation-for-plex/p7 or there is some source here https://github.com/ziggimon/OpenELEC.tv

sjorge, 2015-08-28:
I was talking more about a fresher OpenELEC than 3.2 that currently exists. Seems OpenELEC + PHT looks pretty dead. Swapping the Rasplex PHT from the normal PHT build would probably also work fine to get best of both words.

NedtheNerd, 2015-09-11:
Not really a Rasplex issue, closing.

closed 2015-09-11 by NedtheNerd