RasPlex/RasPlex #421: OpenVPN TLS Error

benjaminwolf, 2015-08-31:
Looks like this is a Rasplex issue. The following test was done using Raspberry Pi 2 with the same client.ovpn file, on the same network (using same IP even) attempting to connect to the same OpenVPN server.

Using OpenElec 5.0.8, I was able to SSH and connect OpenVPN successfully using the same client config file. Rasplex 0.7.0 gives a TLS error.

OpenElec 5.0.8
```
OpenELEC (official) Version: 5.0.8
OpenELEC:~ # openvpn /storage/vpnconfig/client.ovpn
Sat Aug 22 22:30:14 2015 OpenVPN 2.3.5 armv7ve-openelec-linux-gnueabi [SSL (OpenSSL)] [LZO] [EPOLL] [IPv6] built on Mar 31 2015
Sat Aug 22 22:30:14 2015 library versions: LibreSSL 2.1, LZO 2.08
Sat Aug 22 22:30:14 2015 WARNING: file '/storage/vpnconfig/streetcred.txt' is group or others accessible
Sat Aug 22 22:30:14 2015 UDPv4 link local: [undef]
Sat Aug 22 22:30:14 2015 UDPv4 link remote: [AF_INET]MY_ROUTER_IP_ADDRESS:1194
Sat Aug 22 22:30:14 2015 WARNING: this configuration may cache passwords in memory -- use the auth-nocache option to prevent this
Sat Aug 22 22:30:21 2015 [RT-AC68U] Peer Connection Initiated with [AF_INET]MY_ROUTER_IP_ADDRESS:1194
Sat Aug 22 22:30:23 2015 TUN/TAP device tun0 opened
Sat Aug 22 22:30:23 2015 do_ifconfig, tt->ipv6=0, tt->did_ifconfig_ipv6_setup=0
Sat Aug 22 22:30:23 2015 /sbin/ip link set dev tun0 up mtu 1500
Sat Aug 22 22:30:23 2015 /sbin/ip addr add dev tun0 local 10.8.0.10 peer 10.8.0.9
Sat Aug 22 22:30:23 2015 Initialization Sequence Completed
^Z[1]+ Stopped openvpn /storage/vpnconfig/client.ovpn
```
And the Rasplex Error below:

Rasplex 0.7.0
```
RasPlex:~ # openvpn /storage/vpnconfig/client.ovpn
Sat Aug 22 23:02:23 2015 OpenVPN 2.3.7 armv7ve-openelec-linux-gnueabi [SSL (OpenSSL)] [LZO] [EPOLL] [IPv6] built on Jun 16 2015
Sat Aug 22 23:02:23 2015 library versions: LibreSSL 2.1.7, LZO 2.08
Sat Aug 22 23:02:23 2015 WARNING: file '/storage/vpnconfig/streetcred.txt' is group or others accessible
Sat Aug 22 23:02:23 2015 UDPv4 link local: [undef]
Sat Aug 22 23:02:23 2015 UDPv4 link remote: [AF_INET]MY_ROUTER_IP_ADDRESS:1194
Sat Aug 22 23:02:23 2015 WARNING: this configuration may cache passwords in memory -- use the auth-nocache option to prevent this
Sat Aug 22 23:02:23 2015 TLS_ERROR: BIO read tls_read_plaintext error: error:1408D06E:SSL routines:SSL3_GET_KEY_EXCHANGE:bad dh p length
Sat Aug 22 23:02:23 2015 TLS Error: TLS object -> incoming plaintext read error
Sat Aug 22 23:02:23 2015 TLS Error: TLS handshake failed
Sat Aug 22 23:02:23 2015 SIGUSR1[soft,tls-error] received, process restarting
^Z[1]+ Stopped openvpn /storage/vpnconfig/client.ovpn
```
Seems like the OpenVPN portion of the OpenElec wasn't ported correctly, or the Rasplex 0.7.0 is using a version of OpenElec that had a bug in it.

Raspberry Pi CPUINFO
```
Hardware        : BCM2709
Revision        : a01041
Serial          : 00000000bb3a3032
```

Kwiboo, 2015-09-01:
RasPlex 0.7.0 is using a newer version of LibreSSL that includes fixes for the logjam vulnerability.
Your OpenVPN server is probably using an outdated version and/or using a weak Diffie–Hellman key, you should upgrade or replace your Diffie–Hellman key.
See https://blog.cloudflare.com/logjam-the-latest-tls-vulnerability-explained/ for more info about the logjam vulnerability.

Note: RasPlex 0.7.1 will have OpenVPN 2.3.8

benjaminwolf, 2015-09-03:
Thanks, kwiboo. You were spot on. An upgrade to the OpenVPN server and new 1028 bit Diffie-Hellman key sorted me out. Thanks again!

PS. I would say, it would be great if someone in the OpenElec team made a better openVPN GUI. That thing is pretty much unusable.

closed 2015-09-03 by benjaminwolf