RasPlex/RasPlex #424: Issue since ir remote mapping update

cpfaff, 2015-09-11:
He

I have problems with the latest release where you updated the lirc mappings. My remote has a button that reads "subtitles" which was working nicely so far. But since the update it does something different and I do not know exactly what. If a movie is running an I press it the movie vanishes into the background and an empty page appears with a sidebar on the left that I have never ever seen before. How can I get this old button back?

All the best

cpfaff, 2015-09-11:
Seems the default lirc map of RasPlex maps now the subtitle to the key "KEY_SUBTITLE" which is not the right button sent by my remote. My button is named "KEY_GREEN" thus I created now 2 files. 

1) Lircmap.xml in ~/.plexht/userdata/

with the content:

```xml
<lircmap>
   <remote device="devinput">
       <subtitle>KEY_GREEN</subtitle>
   </remote>
</lircmap>
```

2) remote.xml in ~/.plexht/userdate/keymaps

with the content:

```xml
<keymap>
  <global>
    <remote>
      <subtitle>NextSubtitle</subtitle>
    </remote>
  </global>
</keymap>
```

After that I restarted and it works again. Is this a bug? Are the button names standardized in any way?

cpfaff, 2015-09-11:
Is it possible to have multiple possibilities for the right key specified in the default Lircmap.xml. I mean something like

```xml
<lircmap>
   <remote device="devinput">
       <subtitle>KEY_GREEN</subtitle>
       <subtitle>KEY_SUBTITLE</subtitle>
   </remote>
</lircmap>
```

NedtheNerd, 2015-09-11:
Not a bug, the release notes did mention that you may have to change some mapping so.

Closing issue.

closed 2015-09-11 by NedtheNerd