RasPlex/RasPlex #425: [Feature Request] Either force transcode or throw error warning for incompatible audio codec

sashahilton00, 2015-09-11:
Hi,

I have an oldish flatscreen, and for a few months I have consistently had problems playing certain files. As I discovered today, after upgrading to 0.7.1 and still facing issues, the problem lies in the fact that some of my content has a DTS audio track associated with it, whilst my TV can only support 2-channel and AC3. The only way I found this out was stumbling across a document on the internet to check compatibility using <code>tvservice -a</code> after one has ssh'd into the RasPlex instance. (Document here: https://15254b2dcaab7f5478ab-24461f391e20b7336331d5789078af53.ssl.cf1.rackcdn.com/plex.vanillacommunity.com/editor/9d/ip770nuywwik.pdf). Might I suggest that, for incompatible audio codecs, asopposed to showing the loading icon in the bottom left, and never loading, upon pressing play, RasPlex visibly fails and displays a message such as 'The content you are trying to play does not include a compatible audio track. Please ensure your files contain an audio track in at least one of the following formats: [insert compatible formats discovered by <code>tvservice -a</code> here. eg. PCM, AC3]

This way, people will actually realise what the problem is and be able to fix it, as opposed to silent failiure which is a PITA to track down and rectify.

MarshallOfSound, 2015-11-06:
I believe in the settings of RasPlex you can tell it whether or not your TV / player supports DTS audio.  It's just a checkbox under audio :)

NedtheNerd, 2015-11-07:
You need to ensure that you have set the correct audio settings, see this topic https://forums.plex.tv/discussion/181255/rasplex-guide-to-audio-settings

davidjb, 2015-12-06:
I had the same issue and had to follow the same process in discovering how to debug the problem (although this GitHub issue was where I landed and @NedtheNerd's comment helped).  My other issue was that whilst my speakers support DTS, my TV doesn't do passthrough, only forwards AC3 and PCM.  So, +1 that makes audio compatibility more obvious; @sashahilton00's suggestion is good, and I'd suggest some descriptor text on the audio settings preference pane about determining compatibility (or a web link to said PDF) would be even more helpful for new users.

For note, a crafty solution to getting DTS to your speakers is covered by an [OpenELEC forum post](http://openelec.tv/forum/68-audio/43659-how-to-play-dts-movies-with-raspberry-pi-without-stutters-enabling-passthrough).

Nornode, 2016-02-15:
I just want to  make this clear for others that read and research:
Rasplex does not yet support DTS re-routing/passthrough
OSMC - Can do "proper" DTS and sound re-routing/passthrough to an external DAC.
To me it looks like the options are available in Rasplex but as I understood, they are only "active" for the music parts and not for DTS passthrough (sound effects like TV series themes work).

I got an external USB DAC connected to the Raspi to send DTS (all) signals directly to my receiver over toss-link/coax that decodes the DTS and as I said this works on OSMC but not on Rasplex, yet.

I'm really looking forward to have proper DTS passthrough/rerouting feature working on the Raplex!