RasPlex/RasPlex #426: No thumbnail images display in library when skipping to new location - 100% Test Case

edddeduck, 2015-09-11:
Reproduction Steps:

Select a library with a large number of films or TV shows. Make sure you're in default view (mini posters).
Navigate your film (or TV) collection using letters (I did this by pressing the numbers on the TV remote, using CEC) Plex will then jump to the letter you selected. For example press 8 and it will take you to "T".

Expected:
It will take you to position T in the library and display the items starting with T and later.

Actual:
It will take you to position T in the library but doesn't display any thumbnails just the default "Plex" image you get if your thumbs haven't loaded in. This will never update or recover until I press a direction (then it updates everything visible).

100% Reproduction using the above test case instructions using a Pi2 with clean vanilla install or 0.6.0, 0.7.0 and 0.7.1. 

Also seen by https://forums.plex.tv/profile/Rikrok in the Rasplex area on the Plex forums.

Kwiboo, 2015-09-12:
@edddeduck Excellent bug report, very easy to reproduce.
The sms jump, first/last page and next/prev letter actions will refresh thumbnail images correctly in next release.

NedtheNerd, 2015-11-09:
Can you check in latest release 1.0.0 available now and advise.

edddeduck, 2015-11-10:
Tested and verified as fixed in 1.0.0

closed 2015-11-10 by edddeduck