RasPlex/RasPlex #427: RasPlex 0.7.1 Video Studdering

chesterdkat, 2015-09-14:
2 different Raspberry Pi 2's both have the same exact issue where certain videos will first go out of voice sync then after stopping the stream and restarting the voice is now synced but the video will studder. I also have 2 different servers, one is an Atom Dual Core 1.8 Ghz running Debian 7, 4Gigs RAM and the other is a Mac Mini i7 16 Gigs of RAM. This issue is the same regardless of the server being used. Both Raspberry Pi's are connected via Ethernet to Gigabit switches and both servers are also Gigabit Ethernet. Video over the home network is never transcoded but is always direct play.

dsfser, 2015-09-26:
i have the same problem. i have a raspberry pi B+. when i try to play a avi 720p with these specs:

== Size: 2.49GB
== Pars: 6%
== Format: AVI / XViD / 3449 Kbit/s
== Resolution: 1280 X 536 @ 23.976 Fps
== Audio: English AC3 6 CHannels/384kbps
== Subs: Dutch (Ingebakken)

i get the studdering.

i will try the raspberry pi 2B now!

chesterdkat, 2015-09-27:
0.7.0 worked perfectly.

Dave

NedtheNerd, 2015-11-09:
Can you check in latest release 1.0.0 and advise.

NedtheNerd, 2015-11-18:
Closing issue, confirmed as fixed in forums.

closed 2015-11-18 by NedtheNerd