RasPlex/RasPlex #428: CEC stops working after short period of time

mihalski, 2015-09-14:
If you connect to RasPlex with the Plex app as remote then CEC starts working again. This is happening in 0.7.1 and is unchanged from 0.7 which is the first version I have used. I am using the Raspberry Pi 2B.

NedtheNerd, 2015-11-09:
Can you please check in latest release - 1.0.0 - and advise

viperet, 2015-11-18:
I can confirm this issue - for me CEC remote stops working after I turn OFF and back ON my TV (leaving raspberry working). Then, if press some key on Ples iOS app as remote - CEC starts working again. Im on 0.7.1, for now I can't test this issue on 1.0.0 because it is not working for me at all (errors during boot and no network connections)

NedtheNerd, 2015-11-18:
@viperet Are you sure you flashed the right image, there is one for the RPi Model B/B+ and one for the RPi2.  This has been fixed in 1.0.0. Post in the forums your specific issues on booting 1.0.0.

Closing issue as fixed.

closed 2015-11-18 by NedtheNerd