RasPlex/RasPlex #429: Loosing connection and get directory failed error

cpfaff, 2015-09-14:
He

I have problems with the latest RasPlex release 0.7.1. I have a random loss of the connection to the server with no error message visible to the interface. Additionally I sometimes get an error message that reads "Get directory faild" after playing some content and then hit stop. I check if I can find the time to reproduce and log it to send you the logs for investigation. Where are the logs stored by the way? And do I have to turn them on?

All the best.

I was posting about my problems here as well: 

http://forums.plex.tv/discussion/comment/1018260#Comment_1018260

cpfaff, 2015-09-14:
Collection of information:

* serial: (will follow)
* crash ID (?)
* size of PI (Pi 2/1GB)
* SD card (SDHC class 10)
* connection type (wifi)
* RasPlex version (0.7.1)
* Lofiles (follow)
* install system/method (originally from Linux/ but now internal update from 0.7.0)

cpfaff, 2015-09-14:
Currently I am not able to reproduce the error to get a logfile. I will leave debug logging on and upload the logs when I get something.

cpfaff, 2015-09-18:
I got it again now. Updated the post above with the link to the pastebin for the log file.

Kwiboo, 2015-09-18:
@cpfaff Need complete log file not just lines containing error.

cpfaff, 2015-09-18:
ok fine. Just a Moment

cpfaff, 2015-09-18:
https://www.dropbox.com/s/j6t6n3022wv219g/plexhometheater.log?dl=0

cpfaff, 2015-09-18:
By the way is it secure to share a full log file of rasplex? I mean just because of the token based authentication thing. Is everything filtered out from the log?

Kwiboo, 2015-09-18:
The token should be filtered out, but I have downloaded the log now, so you can remove it, thanks.

Kwiboo, 2015-09-18:
From the logs it looks like there is a real issue with the PMS connection handling, after some time RasPlex for some reason wrongly sees the PMS as departing and because of this the local connection is removed and only the external non working connection is left.
After some time the servers is refreshed from plex.tv or GDM and then the local connection is re-added.

Thinks you can try:
- Disable GDM in PMS and reboot RasPlex
- See if you can enable reverse NAT so you can access your PMS on your external IP from your internal network

cpfaff, 2015-09-18:
OK. Thanks. I will try it. As this is a problem with the server do we report it to the plex forum in the bug section? By the way would be setting the local server address manually in rasplex also an option?

cpfaff, 2015-09-18:
Ok thanks I will try.

Kwiboo, 2015-09-18:
It is most likely an issue with RasPlex's handling of the connections, and not a PMS bug.

RasPlex keeps track of each server and its associated connection (local ip, external ip) and they can come from GDM (multicast discovery) or from plex.tv.
When RasPlex received a departing signal it should not have removed the connection but instead flag it as unreachable. And later tried to reconnect.

RasPlex contains some modification to the connection handling compared to regular PHT (using ssl-connection even when not signed in, merging non-ssl with ssl connection if they point to same IP), possible we have introduced new bugs not seen in PHT because of the changes.

NedtheNerd, 2015-11-09:
Should be fixed in 1.0.0, available now, please check and advise.

cpfaff, 2015-11-10:
Cool. Currently I switched to the preview of plex media player and cannot test it. Will switch back on the following weekend and report back then.

NedtheNerd, 2015-11-11:
Reported as fixed in the forums in release 1.0.0, closing issue.

closed 2015-11-11 by NedtheNerd