RasPlex/RasPlex #430: RasPlex Crashes intermittently on various media

tonyp1983, 2015-09-23:
My freshly installed copy of RasPlex intermittently locks up, stopping video and audio and making it impossible to access via ssh. It is connected to the network via gigabit ethernet directly into the router. 

On first installation last night this happened during the play of a 720p movie after 45 minutes and then intermittently for a while. I reflashed the SD card and played again for a while with a couple of hiccups. Today whilst playing an episode of a TV series it played fine for 30 minutes and then locked up.

Never had any issues with OpenELEC which I've used for about 18 months. One thing that's concerning is that the log file indicates an attempt to connect to a private IP in the range 192.168.0.x, my home network is all on 10.0.0.x.

Crash ID: Not known
Size of Raspberry Pi (256, 512): 512
Type of SD card (SDHC class 10?) SDHC class 10 16 GB
Internet connection type (Ethernet or Wifi) Ethernet
Version of RasPlex (if not the latest, please see if the problem exists on latest first): 0.71
Log files (ensure that you enable "debug logging in Preferences -> System -> advanced" and try to reproduce the issue): https://www.dropbox.com/s/x4pez4v42vfp7x4/plexhometheater.old.log?dl=0
How did you install rasplex? (gui installer, win32 disk imager, what platform OSX/Win/Linux?) OS X gui installer
Best description you can give of the problem: See above
Steps to reproduce (what were you doing when it happened? Can you make it happen again?): Watching random videos of various sizes, resolutions.

PMS is the latest version for FreeBSD: 0.9.12.12

Kwiboo, 2015-09-24:
@tonyp1983 the log file do not contain any debug logging, please enable debug logging in preferences and try to recreate the issue.
What type of PSU are you using and what is the quality of the usb cable?
What type of codec, bitrate etc are you having problems with?

tonyp1983, 2015-09-28:
@Kwiboo My apologies, clearly can't follow instructions either. Here's a link to the latest log files with debug logging enabled https://dl.dropboxusercontent.com/u/8209178/log-2015-09-28-21.42.29.zip
I'm using an official iPad plug with a good quality usb cable (I've used it for the last couple of months with OpenELEC with absolutely no issues). The codecs and bitrates have varied greatly from low bitrate xvid to full 1080p with high bitrates, however I've been away over the weekend and not even used RasPlex and the unit had locked up when I got back.

Kwiboo, 2015-09-29:
Thanks for the logs. I have only found one suspicious line in the logfile, `11:50:09 T:1486877776   DEBUG: ffmpeg[589FF450]: [https] HTTP error 416 Requested Range Not Satisfiable`.
Try to disable FFMPEG AVIO in Preferences and see if that helps.

tonyp1983, 2015-09-29:
Thanks for the quick response. I've disabled FFMPEG AVIO, I'll see how it goes over the next few days and report back.

Kwiboo, 2015-09-29:
I can also see that the requests to your PMS is failing shortly after, `11:53:14 T:1486877776 WARNING: CCurlFile::FillBuffer [http://10.0.0.20:32500/?X-Plex-Token=SECRETSTUFF] failed: Timeout was reached`.
Could it be that your PMS crashed or some other network problem occurred around that time?
Check the log files on your PMS around that time and see if there are anything interesting there.

Kwiboo, 2015-09-29:
Wrong of me, that is probably your phone or similar that always seems to fail with a timeout error.
But still interesting to see if your PMS reports anything around 11:50:09 when the HTTP error occurred.

tonyp1983, 2015-09-29:
Well there's something that looks very strange to me (not that I know what it means) in the PMS logs from earlier in the day yesterday and went on for quite some time.
https://dl.dropboxusercontent.com/u/8209178/Plex%20Media%20Server.log

NedtheNerd, 2015-11-09:
There are some changes in 1.0.0, now available, can you check on this version.

aselwyn1, 2015-12-07:
i just started using the 1.0.1 version of rasplex and im getting the same error. im just starting to try rasplex from xbmc that i used for a while for media so just setup a server for plex with some mp3 files and a single mp4 video. today the video works fine playing but not the mp3 music files. a 3 days ago when i originally installed rasplex it was working for those mp3's. im running a model b with ethernet with a proper psu from Newark for the pi.

g1t-dlanor, 2015-12-08:
@aselwyn1 : 
If it consistently worked fine 3 days ago, and consistently fails now, then something must have changed in between these tests. If that change is not located in your network or your PMS server, then I suspect a corrupted SD card, which can occur for various reasons, the most common being a low class SD card (any class lower than 10) or a weak PSU. Please double-check your equipment to make sure the PSU can deliver sufficient current to deal with peak loads (2+ Amp for an RPi2, 1+ Amp for an RPi1 are preferred PSU ratings, but you may need more if you use a WiFi dongle). Also check that your SD card is marked as Class 10 or UHS-1. (Capital "C" with the number 10 inscribed or capital "U" with the number 1 inscribed.)

Please also test if a newly flashed RasPlex card can give the same results you had three days ago.
And if that doesn't work, I think it indicates some failure outside of the RPi, possibly to do with the configuration of your brand new PMS server.

aselwyn1, 2015-12-09:
as i said in my first comment i know the psu is known good and yes the sd card a high quality class 10 as its worked for about a year already on xbmc. i went back and tried it today and it works just fine so i don't even know what happend

NedtheNerd, 2015-12-09:
Closing issue.

closed 2015-12-09 by NedtheNerd