RasPlex/RasPlex #431: plex account login faled

new-m, 2015-10-03:
Hello, I have this:
RasPlex (official) Version: RP-0.7.1
RasPlex git: b5fd0f434b7bf803ac4f26914ced9ec8c6717d7c
And I cannot login to my plex pass account. Only working with local media server.
When I trying to login page is freezing on PIN load. If I using email and password - it's freezing too.
Same issue was published heare: https://github.com/RasPlex/RasPlex/issues/379
How I can help to troubleshoot this?

new-m, 2015-10-03:
Also in debug log I have this problem:
22:03:56 T:1778103376   DEBUG: CurlFile::Open(0x5b64deb8) https://10-230-230-113.1aa535db6a6045a8b101b4705b59ebb7.plex.direct:32400/library/sections
22:03:56 T:1803269200 WARNING: CCurlFile::FillBuffer [https://plex.tv:443/pins.xml] failed: Timeout was reached
22:03:56 T:1803269200   DEBUG: CCurlFile::Service reading extra failure information
22:03:56 T:1803269200   ERROR: CMyPlexManager::GetXml failed to fetch plexserver://myplex/pins.xml : -1
22:03:56 T:1803269200   DEBUG: CMyPlexManager::BroadcastState state: refresh, error: network
22:03:56 T:1803269200   DEBUG: CMyPlexManager::Process after a run our state is 0, will now sleep for 3600 seconds
22:03:56 T:1778103376   DEBUG: CPlexDirectory::GetDirectory::Timing took 0.924000 seconds to download XML document

https://forums.plex.tv/discussion/181944/raspberry-pi-2-running-0-7-0-or-0-7-1-will-not-sign-in

NedtheNerd, 2015-11-09:
Can you load up 1.0.0 and see if your issue is solved.

NedtheNerd, 2015-11-18:
No further feedback, closing issue.

closed 2015-11-18 by NedtheNerd