RasPlex/RasPlex #433: 0.7.0 and 0.7.1 locks on initial Plex for OPENELEC screen

vger74656, 2015-10-21:
Hi all,

As the title suggests, flash 0.7.0 or 0.7.1 onto a new SD card and when turned on it doens't get passed the initial splash screen.  I have also tried 0.6.0 and that one works fine.

Thoughts?  Anything you need (logs)?

vger74656, 2015-10-21:
Should also mention I am running RPi2

g1t-dlanor, 2015-10-21:
Either you are mistaken or else you have some problem specific to some hardware at your end, as I always test each version, both for RPi1 and RPi2, by flashing to SD card and booting clean to configure 'from scratch', and that has worked for me with both 0.7.0 and 0.7.1. So if it really fails for you then it must be due to some difference in our hardware.

Hopefully you have just mistaken the VERY long boot delay in the very first boot session, believing that it was a boot failure. This long delay for the very first boot is due to the need to extend an EXT4 partition to the true card size and then unpack the OpenELEC Linux operating system files from an archive file on the FAT32 partition into the new EXT4 partition. This inevitably takes some time to do and that time has grown in recent versions compared to older ones (due to system improvements!).

Please try again, and make sure to give it a minute or two to complete that first boot.
If it still fails for you then we need to examine what differencies exist between your setup and those that don't have your problem. (Like quality of SD card, Power Supply, Power cables etc)

vger74656, 2015-10-21:
Hi Ulf,

I used your installer about 3 times and then tried the manual method to
install the 0.7.1 Pi2 image.  I also tried the Pi image but that didn't
even bring up the load screen (just a screen of color).  Upon booting on
the first attempt, I did wait for about 20 minutes before giving up, and
there was no text on the top of the screen advising what the installer was
doing, like there is in 0.6.0.

One thing I did do was to try to manually update the version from 0.6.0 to
0.7.1 by just copying the SYSTEM file and all other files in the upgrade
zip file to their relevant locations, but it complained about a modified
config file and that the install was broken, wait 60 seconds, then ran the
new version of Plex, but there were no display drivers installed, 640x480
and very pixelated.

I also tried 3 USB cables, plugged directly into the back of the TV, and
into 2 different power packs, one providing 2A if it needed it.  I also
have a USB current meter that tells me the draw on the USB port and it
didn't go above 0.26A 5.1V during boot of both 0.6.0 and 0.7.1 so I don't
think it's a draw related issue.

The thing I can't get passed is that the Pi2 works perfectly on 0.6.0.  If
there was a current/load issue or a hardware fault with my Pi2, wouldn't it
manifest in 0.6.0, not just 0.7.0/0.7.1?

If there is anything else I can provide for you to assist with this, let me
know.

Cheers,

Glen

g1t-dlanor, 2015-10-22:
It does seem strange that you had no trouble with an older version, and still can't use recent versions at all. We'll just have to dig a little deeper in our attempts to figure out why.

Your method of trying to manually update 0.6.0  was probably not valid, as an update from an older version needs to be done in a very specific way, and you probably didn't get  it right.

Your 2A PSU should be sufficient, and that is what I recommend for you to use.
TV USB outlets are NOT recommended, not only because their current capacity is doubtful, but also because turning off the TV set may cause SD card corruption if it occurs during SD card access by RasPlex. (CEC protocol may help in avoiding such corruption, but that depends on TV set implementation of CEC.)
As for the measured current load, the average value is not very relevant, as the CPU may crash if the PSU is unable to deliver a momentary peak current spike. These may be very short in duration, and will probably not be visible on a current meter (unless it is capable of tracing peaks for later inspection). The same goes for watching a voltage meter for drops. Momentary voltage drops will not be visible on an averaging meter, but are sure to crash the CPU if the voltage goes much below 5 volts (and here half a volt is VERY much).

One thing you haven't mentioned is the quality of your SD card.
Please verify that the card you use is either SD class 10 or UHS-1.
This is normally shown by a small logo consisting of a capital "C" with the number "10" inside the letter (in smaller font of course), or a capital "U" with the number "1" similarily inscribed.
Lower class cards are known to be unstable with RasPlex, though it may work for some versions.

Another thing you've not mentioned is whether or not you've tried the internal update method of RasPlex, which is what I recommend that you try next. I've just performed an experiment myself with a freshly flashed and configured RasPlex 0.6.0, using the internal update method to turn it into RasPlex 0.7.1, and this seems to have worked fine. So hopefully it will work for you as well.

For this you need to go into the submenu: Preferences > System >  Software Update
There you need to switch the channel selection from the default "Stable" to "PreRelease".
(Although I consider 0.7.1 to be very stable, it's still classified as "PreRelease".)
If you have auto-update enabled this should cause an automatic check for 0.7.1, but if you have auto-update disabled (I always do myself, as I hate unrequested updating), then you will need to use the command "Check for update now..." in that same submenu.

Unlike the other submenu entries that's not a setting, but a directly executed command, which when an update is found also leads to additional interactive responses for performing the update (which uses update files downloaded at the time). Just follow the instructions and be patient during the reboots that follow (required as system files can not be replaced while the fully booted system is running).

If that update method also fails then there must be something in the new versions which is incompatible with your hardware setup. (Possibly the new and improved CEC functions interact badly with your TV during boot, which can cause black screen even if RasPlex has booted, in case it uses a video mode incompatible to the TV.) But hopefully that will not be the case.

Btw: Though my full name is "Ulf Ronald Andersson", I'm normally called by my middle name "Ronald" (there was also another "Ulf" in the family, so using that name would cause mixups).

Best regards: dlanor (my username in most forums, though sometimes with additional prefixes)

vger74656, 2015-10-22:
Hi Ronald (apologies :P),

I am using a Class 10 Kingston micro SD card bought from a retail outlet
(not fake like the ones on eBay).  I have not tried doing an internal
update, of which I can do tonight when I get home from work.

Once I have tried that, I will contact and we can take it from there.

Thanks for your time and assistance :)

Cheers,

Glen

vger74656, 2015-10-23:
Hi Ronald,

Just did the upgrade from 0.6.0 to 0.7.1 via the Rasplex update interface
and that worked perfectly.  Is there any systemic difference in the way it
leaves the installation from 0.6.0 rather than just flashing with 0.7.1?

Cheers,

Glen

g1t-dlanor, 2015-10-24:
I'm not entirely sure about any difference in results from these install methods.
The one thing I'd suspect possible is a difference in updated contents of the boot card, such as the optional device addon modules in the "overlays" folder. But for many users this may be irrelevant.
(You don't need modules for hardware addons you don't have.)

Perhaps one of the other guys in the project can chime in with more details on exactly how perfectly (or imperfectly) the internal update code of 0.6.0 can handle an update to 0.7.1.

NedtheNerd, 2015-11-09:
Not an issue, specific to only this user, have now tested every release from 0.6.0 to 1.0.0 without any problems, closing issue.

closed 2015-11-09 by NedtheNerd