RasPlex/RasPlex #434: Managed users can't reboot/shutdown RasPlex 0.7.1

bbaraniec, 2015-11-01:
I have updated my Rasberry Pi 2 B to 0.7.1. When you press left arrow on main menu you have options like reboot, shutdown and switch user. Only admin of Home Plex has option to reboot and shutdown, all managed users have switch user only. Is this intended? PHT on Windows works for them.
As requested by NedtheNerd https://forums.plex.tv/discussion/comment/1049348/#Comment_1049348

Kwiboo, 2015-12-02:
In PHT on Windows/OSX Quit and Sleep is available for managed users and Shut Down is restricted to admins only. On RasPlex the Quit and Sleep function is removed since there are no equivalent functionality supported but a Restart option has been added for admins only.
Next release will have a skin option to enable Shut Down and Restart for managed users.

bbaraniec, 2015-12-02:
Great. Thanks for that.

NedtheNerd, 2016-01-19:
Fixed in next release - 1.0.2.

closed 2016-01-19 by NedtheNerd

bmlevine11, 2016-01-19:
Hey

bmlevine11, 2016-01-19:
Join me