RasPlex/RasPlex #435: Video Calibration not saving to /storage/.plexht/userdata/guisettings.xml

pkinnane, 2015-11-02:
Hi - if I change Preferences->Advanced->Video Calibration, when I next reboot the calibration is returned to default values.  This appears to be the same problem documented at http://forum.kodi.tv/showthread.php?tid=135250.

Thanks

Paul

pkinnane, 2015-11-02:
0.7.1
Raspberry Pi 2b
8GB class 10 sd card
Wifi
Installed via win32 disk imager

NedtheNerd, 2015-11-09:
Checked this in Rasplex 1.0.0, calibration values are saved and persist on next boot, closing issue.

pkinnane, 2015-11-10:
I updated from 0.7.1 to 1.0.0 and the problem persists.  Calibration settings are being lost after rebooting.

pkinnane, 2015-11-10:
Did a clean install of 1.0.0 and issue appears to be fixed there.  However I now cannot set a static IP address.  After entering the static IP address there is no way to highlight the Save button (using Swift).

NedtheNerd, 2015-11-10:
Can you please start a new issue for this, closing original issue.

closed 2015-11-10 by NedtheNerd