RasPlex/RasPlex #436: 5.1 downmix to stereo clips audio in 1.0.0

jishi, 2015-11-08:
In 0.7.1 I felt that the option "boost audio on downmix" was sort of reversed. With that option enabled, all videos with 5.1 audio seemed to have substantionally lower volumes than 2.0 videos. When unchecking that option, it seemed to adjust itself to be more normal (like previous version), so I didn't think about it much after that.

Now, in the 1.0.0 release that I installed today, this has become much worse. Now, if the "boost" setting is unchecked, 5.1 tracks will actually be insanely boosted, and even clip horribly. Checking "boost audio on downmix", seems to lower the output, within non-clipping range, which seems very counter intuitive. Another weird thing which is probably related, now all intro music etc (that probably has a right-left dominance in the 5.1 track) is much, much lower than before, which indicates that the center channel seem to be boosted the most. I have been fiddling with the other setting "boost center channel", but it doesn't seem to make any difference. 

Adjusting the volume in rasplex to around -6 dB, seems to mitigate the clipping on the few videos I have tested (when boost is deactivated), which I think indicates that it is rasplex itself that adds the additional boost.

I didn't know what kind of info that would actually be useful here, since there is no crash or similar that would be relevant. I can provide sample videos if needed, just let me know where to send them, or I can host them somewhere. Also let me know if a debug log when starting playback would be helpful. 

I'm sending audio over hdmi directly into the TV, and have selected 2.0 output in rasplex preferences.

g1t-dlanor, 2015-11-09:
I have tested the problem you describe and I think no special samples should be needed. The bug is easily triggered using any normal video with 5.1 audio tracks. I've made sure that others of the RasPlex team are also aware of it, so hopefully this will be fixed ASAP.

attaxia, 2015-11-20:
Just wanted to chime in and report the same issue.

joshstrange, 2015-11-23:
Oh thank goodness, I thought I was going crazy as I didn't notice it close to the upgrade so I didn't link the two. I changed out HDMI cables, bypassed the switcher, tried TV audio instead of the soundbar. It's not really noticeable in most cases but the two newest Doctor Who episodes must use those lower bands more because I had subtitles on and I could really notice missing dialog. I've switched over to my FireTV for now till this gets fixed, hopefully it's not too much longer as I love RasPlex.

g1t-dlanor, 2015-11-23:
This issue is fixed in the current beta test version, so a fixed version will very soon be released to the public.

NedtheNerd, 2015-12-01:
Fixed in 1.0.1 - now released, closing issue.

closed 2015-12-01 by NedtheNerd