RasPlex/RasPlex #437: Segmentation Fault when USB-CEC adapter attached

ElliottH, 2015-11-11:
I have a [Pulse Eight USB-CEC adapter](https://www.pulse-eight.com/p/104/usb-hdmi-cec-adapter) attached to my Pi 2, and since the performing the update I was offered today RasPlex immediately crashes at the splash screen.

Running it under gdb I see:
```
Program received signal SIGSEGV, Segmentation fault.
[Switching to LWP 5900]
0x7591ccac in readdir () from /lib/libc.so.6
(gdb) bt
#0  0x7591ccac in readdir () from /lib/libc.so.6
#1  0x713e2f40 in PLATFORM::CDRMEdidParser::GetPhysicalAddress() () from /usr/lib/libcec.so.3.0.1
#2  0x713d7d38 in CEC::CUSBCECAdapterCommunication::GetPhysicalAddress() () from /usr/lib/libcec.so.3.0.1
#3  0x713d064c in CEC::CCECClient::AutodetectPhysicalAddress() () from /usr/lib/libcec.so.3.0.1
#4  0x713cf5e0 in CEC::CCECClient::SetPhysicalAddress(CEC::libcec_configuration const&) () from /usr/lib/libcec.so.3.0.1
#5  0x713cf968 in CEC::CCECClient::OnRegister() () from /usr/lib/libcec.so.3.0.1
#6  0x713d61d4 in CEC::CCECProcessor::RegisterClient(std::shared_ptr<CEC::CCECClient>) () from /usr/lib/libcec.so.3.0.1
#7  0x713dc254 in CEC::CLibCEC::Open(char const*, unsigned int) () from /usr/lib/libcec.so.3.0.1
#8  0x00b5ebbc in PERIPHERALS::CPeripheralCecAdapter::OpenConnection() ()
#9  0x00b5eddc in PERIPHERALS::CPeripheralCecAdapter::Process() ()
#10 0x00839064 in CThread::Action() ()
#11 0x00839330 in CThread::staticThread(void*) ()
#12 0x75c4ed58 in ?? () from /lib/libpthread.so.0
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

This looks to be the same problem mentioned in [this Debug bug (798190)](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=798190), caused by libcec assuming that `/sys/class/drm` exists.

I can boot the Pi fine without the CEC adapter attached.

I can provide a core dump or a log file if required, but the log file doesn't show any errors.

NedtheNerd, 2015-11-11:
Rasplex has CEC built in, if you are using an HDMI lead to connect to your TV you don't need the adapter.

ElliottH, 2015-11-11:
I'm aware that the Pi is CEC capable, but unfortunately the HDMI switch the Pi is attached to doesn't pass the CEC messages through to the TV correctly, so I have to use the CEC adapter to allow me to bypass the switch for the CEC commands to get my remote to work. i.e.:

```
          USB
----------------------------
|                          |
Pi ------ Switch ------ Adapter ------ TV
    HDMI          HDMI           HDMI
```

It has always worked before the update today without any problems and it now doesn't.

NedtheNerd, 2015-11-12:
Looks like the same issue, there was a kernel and firmware update in the release, no /sys/class/dorm directory is present in Rasplex. libcec3 appears to have been fixed in September, will take a while for it to filter down, hopefully by next Rasplex release

NedtheNerd, 2015-12-01:
Should be fixed in 1.0.1 - now released, can you check and advise.

ElliottH, 2015-12-01:
Just updated, Plex no longer crashes and the CEC device works! Thanks for you help and for a quick turnaround time, much appreciated.

NedtheNerd, 2015-12-01:
Closing issue.

closed 2015-12-01 by NedtheNerd