RasPlex/RasPlex #439: 3.5mm audio issue after 1.0 upgrade

ccomly, 2015-11-13:
So after the upgrade (from 7.0 i think), Some audio can barely be heard, namely music/action sounds. while voice audio is very loud and clear. its almost like a reversed issue from the past when you couldn't hear any voice audio  (it was very quiet)  but everything else was super loud.

Something that was fixed in past releases where 5.1 video's audio would be really really low even on max volume and speaker volume to max. This was fixed by adding 

 <headroom>22</headroom>
 <ac3downmixgain>22.0</ac3downmixgain>

to .plexht/userdata/advancedsettings.xml

and just turn down volume on anything that wasnt 5.1.

Maybe not related but thought i would mention it.

Iv tried messing with the audio settings to get the sound normalised, but it either doesn't play at all on 5.1 videos or what it does now high voice audio, very low everything else audio.

tried reinstalling 7.1, same issue remains. Im considering moving to an even older release like 0.6 to see if that fixes it.

its a strange issue i dont know whats going on.

NedtheNerd, 2015-11-13:
Issue is known, thanks.

ccomly, 2015-11-13:
is there any workaround  available?

g1t-dlanor, 2015-11-14:
No workaround is known.
The downmix code is broken and will have to be fixed in a new RasPlex version.
Unfortunately the resulting sound is almost normal for many cases, which is why this error was not identified at a much earlier stage. Had the error been more glaring we could have caught it directly, so it would never have been released.

ccomly, 2015-11-14:
ah, well for now 0.6 works

NedtheNerd, 2015-11-18:
Similar issue to #436

NedtheNerd, 2015-12-01:
Fixed in 1.0.1 - now released, closing issue.

closed 2015-12-01 by NedtheNerd