RasPlex/RasPlex #440: [Feature Request] Show libraries from specific server

cbjerg, 2015-11-16:
Would be nice with an option to only show libraries from specified server. 
I have 2 Plex Media Servers running on my LAN. One I share with friends, and one is for Plex Home users only. They more or less have the same libraries. In PMP and iOS app I can choose which server I want to show the libraries for, making the home screen a little less cluttered.

dalehamel, 2015-11-16:
Make specific accounts?

g1t-dlanor, 2015-11-16:
Two PMS servers on the same LAN should always have separate Plex accounts.

But when accessed from within the LAN (as with a local RasPlex player) the PMS servers will both respond anyway, unless you change a default setting of the PMS servers, so that they require authentication even on the local network. In old PMS versions there was a special option for this, making it easy to activate or disable. But newer PMS versions instead use a list of networks that define which networks should be allowed access without authentication, and by default that list includes your entire LAN. What you need to do is to modify the string containing this list, so that it only allows unauthenticated access from the PMS computer itself, possibly adding also a specific other computer from where you need to use the Plex/Web administrative interface.

For example, I too have two PMS servers in my LAN, each with a separate account, and for each of these servers I have limited unauthenticated access to the PMS computer, as follows:

1: Start the Plex/Web client in your Web browser (in Windows, D-click the Plex icon in the taskbar 'tray')
2: Click the tools icon near top right corner of the client window to open the settings menu
3: Click the  "Server" entry in the top row of the settings form
4: Click the "Show Advanced" button in the second row of the settings form
5: Click the "Network" entry in the left column of setting categories for the server
6: In the list of options on the right, look for the entry "List of networks that are allowed without auth"
7: Edit that string to be only "127.0.0.1/255.255.255.255" (without the quote characters)
This setting will prevent any other computer/device from accessing this server without proper sign in procedure. (Normally only needed once per client/device as it is remembered for future sessions).

I'm not sure exactly how this will work for 'Plex Home' usage, as I never use that myself, so perhaps you should avoid making this change on the server intended only for local 'Home users', but doing this on the server intended for external friend access should prevent it from being accessed by any Plex client not explicitly signed in.

So your 'Plex Home' users on the same LAN will only see the server intended for them, and for clients accessing from outside your LAN this issue doesn't apply, as they should always see only the server to which they are logged in, or if they have their own plex accounts they will only see content from other serves when it is explicitly shared to them. Either way the problem you describe should no longer exist.

chadberg, 2017-04-09:
Any follow up on this enhancement? I use Plex Cloud as a backup to my primary server, and do not need or want RasPlez to show both my server and Plex Cloud libraries at the same time.