RasPlex/RasPlex #441: Lightberry HD late 2015 - Hyperion

Yoda6, 2015-11-20:
Lightberry HD is working since Rasplex 0.6.0 out of the box (thanks to the integration of hyperion), but since late 2015 (ocotober), the lightberry team switched the led type from "ws2801" to "apa102".

In order to work with this led type, hyperion must be updated. The lightberry team has prodvided the updated version of hyperion [here](https://lightberry.eu/support/updating-hyperion-and-firmware-for-apa102-usb/).

But the version of hyperion cannot be updated on Rasplex 1.0, because the binaries are located in /usr/bin, on the SquashFS, read only, partition. 

Is it possible to update the hyperion version of /usr/bin when you're building the image ?

More Information : I've included an hyperion config file ([hyperion.config.txt](https://github.com/RasPlex/RasPlex/files/40571/hyperion.config.txt), change the extension to json), working with the OpenElec version of the lightberry website ([here](http://img.lightberry.eu/download/OpenELEC-RPi2.arm-lightberry-6.0.0.apa102usbfix.zip)) on a raspberry pi 2

edddeduck, 2015-11-20:
https://github.com/tvdzwan/hyperion - The latest version of the Hyperion project also appears to have a fix for these new LEDs if it's easier to merge in the updated version from the main source instead of via the Lightberry link.

dalehamel, 2015-11-20:
Shouldn't be hard to rebuild

Yoda6, 2015-11-23:
The apa102 feature has not been merged yet to the master branch : https://github.com/tvdzwan/hyperion/pull/414

Kwiboo, 2015-12-01:
Rasplex 1.0.1 has part of tvdzwan/hyperion#414 included but is unfortunately missing the performance fixes commit https://github.com/LightberryEu/hyperion/commit/7ebb72f503ad74a7768deedf648b9e84c0a7cf3c, will be included in next release.

Yoda6, 2015-12-03:
My Lightberry setup (apa102 leds) is working with Rasplex 1.0.1 (I just had to change the hyperion json config file) :) 
I haven't looked at the CPU usage, but on a Raspberry 2 I have no lag.

NedtheNerd, 2015-12-03:
Closing issue.

closed 2015-12-03 by NedtheNerd

Yoda6, 2016-01-13:
FYR, the apa102 feature has been merged in hyperion's master branch : tvdzwan/hyperion#414