RasPlex/RasPlex #443: 5.1 downmix to stereo gives uneven mix

janpeeters70, 2015-11-29:
_I read the thread about distorted sound when boost is deactivated but this is another issue._

I've noticed that a downmix from 5.1 to 2.0 results in an uneven mix. 
E.g. When I view a movie where a voice is speaking in the center and music is in the background on the side channels I hear the music too soft compared to the voice. I've tested this with VLC and Quicktime om a Macbook and both produce an even and balanced mix of speech and music. RasPlex latest version doesn't.
All options in audio hardware settings are set to off except Boost Volume. I'm outputting audio via HDMI to my (2.0) television.

This issue has also been reported here: https://forums.plex.tv/discussion/195417/rasplex-audio-problem#latest  
Chrisnijdam reports: Since i installed Rasplex 1.0 i'm struggeling with an audio problem.
When playing soms files (all MKV's) in certain scenes the music gets really low and the soud effects are really loud. For example when i watch southpark and the theme song comes up, the music is low and the singing voice is very hard.

I also have searched Github issues in vain, so either NedtheNerd thinks that it's the same problem as the audio boost problem but this is really something else. Thanks.

NedtheNerd, 2015-11-29:
Same issue, boost volume is also faulty, fix coming in next release - 1.0.1, fully tested and awaiting release.

janpeeters70, 2015-11-29:
Terrific news, thanks for letting me know that this is part of the same issue. Can't wait for 1.0.1 ;-)

closed 2015-11-29 by janpeeters70