RasPlex/RasPlex #445: shared channels in rasplex

firsttris, 2015-12-02:
Hello,

on a shared libraries i doesn't see "channels". i only see movies und tvshow.
i see channels on the web and on the app but not on rasplex.
tested 1.0.0 and 1.0.1.

have i missed something?

when im logged-in as account owner i see channels why not on a shared library?

regards
firsttris

NedtheNerd, 2015-12-02:
Shared Libraries are just that, Channels are not Libraries. There is no option in PMS to share Channels when sharing Libraries.

firsttris, 2015-12-02:
Sure.. You can share channels on PMS.... 
You just need to active this option:
https://support.plex.tv/hc/en-us/articles/201812473-Why-don-t-Channels-appear-for-shared-users-

firsttris, 2015-12-02:
found out its working if i add a user to plex home. before i added as friend.

Kwiboo, 2015-12-02:
Rasplex currently do not load channels for shared servers. Shared channels is a relative new feature of PMS and next Rasplex release will most probably have support for shared channels.

firsttris, 2015-12-02:
thx for your answer i will close this issue.

closed 2015-12-02 by firsttris