RasPlex/RasPlex #446: unable to play from usb audio

firsttris, 2015-12-06:
hello, my usb-dac (nubert no-pro) is not working with rasplex 1.0.1.

i have the same settings like in openelec. (working in openelec)

audio device: ALSA USB AUDIO DAC, SPDIF
passthrough: ALSA USB AUDIO DAC, S/PDIF

no matter witch setting i choose it plays from hdmi.

regards
tristan

NedtheNerd, 2016-01-11:
USB DAC support is only for Music tracks only at the moment, not for movies or TV shows.  See topics in forum. Closing issue.

closed 2016-01-11 by NedtheNerd

OChrisJonesO, 2016-03-05:
Why would this be music only? what difference does it actually make whether the audio track is from a movie/tv show or a song?

NedtheNerd, 2016-03-05:
Music and Video use different players, hoping to get this addressed in 1.6 release.

OChrisJonesO, 2016-03-05:
Any timeline for that to be supported? or is there any workaround possible right now?

Currently stuck using the crappy speakers in my projector over HDMI for audio on tv shows / movies. Music plays fine over my 5.1 speakers however, (tested and forced stereo output to 5.1, all speakers are active). A bit annoying.

Currently setup using a USB soundcard with Logitech z506 speakers. Under RasPlex preferences, audio output device shows up just fine (ALSA:USB Sound Device (@:CARD=Device,DEV=0), Analog).

NedtheNerd, 2016-03-05:
No timeline and no workarounds at the moment, sorry.

Ribwich12, 2016-04-22:
I'm just running into this issue myself. Has there been any progress?

NedtheNerd, 2016-04-22:
Yes, as previously stated, next release.

Ribwich12, 2016-04-22:
Ok thanks. I'm new to Rasplex so I'm not familiar with what release was next.