RasPlex/RasPlex #447: Live stream support

iBobik, 2015-12-10:
It is build on top of Kodi and OpenELEC, so RTSP streams with H264 should be supporter. How to play them (I have URL)? I don't see any UI for it.

g1t-dlanor, 2015-12-10:
There are  lots of things in Kodi which are not supported by any version of 'Plex Home Theater' (which RasPlex is the RPi port of), and direct user entry of video path strings (whether URL or local) is one of them.

With Plex remote streams on Internet sites are normally accessed through 'Channels' which are a Plex-specific form of XBMC/Kodi plugins.

There are some skins which have implemented direct entry of local drive path strings for PHT, and it's possible a similar approach can be used for remote stream links. But to my knowledge this has never been done.

Another approach which probably would work is to design a generic 'Channel' allowing access to user-defined stream links. But this too is something I don't think anyone has done as yet.

iBobik, 2015-12-10:
Thanks for reply.

Channel seems good way, I will try it.

My use case is to be able to show stream from IP cameras on all TVs in the
building.

Jan Pobořil

NedtheNerd, 2015-12-10:
This functionality will only ever be provided by Plex Channels.

Several posts regarding this in the forums.

https://forums.plex.tv/discussion/177711/live-streaming-channel-from-camera-input
https://forums.plex.tv/discussion/175733/get-ip-cameras-as-channels

Closing issue.

closed 2015-12-10 by NedtheNerd