RasPlex/RasPlex #449: HDMI-CEC has stopped working

Wonderpants, 2015-12-16:
I am using a Samsung 55" LED 1080p Smart 3D HDTV (UN55F8000) with the touch remote. It has been functioning flawlessly for about 1-2 months. Recently, within the last week or so, it has stopped loading. In TV Settings HDMI-CEC is still enabled but it is no longer registering when I flip over to my R-Pi/RasPlex system.

I've done the normal things. Reboot/relaunch RPi. Reboot TV, turn off HDMICEC in TV and then turn it back on. Change inputs. Still having issue with this functioning.

Any ideas would be great. Thanks

Wonderpants, 2015-12-16:
To give further information, Version info is 
"Plex 2.4.27"
"Plex Media Player" 1.0.0.4"
Linux Raspberry Pi OpenELEC.

I have my TV fully up to date but I have had it set to manual updates only and had not updated my TV until this wasn't working. I then updated my TV as a troubleshooting step.

NedtheNerd, 2015-12-17:
I think you have the wrong application, from the version info you provided you are running Plex Media Player not Rasplex, suggest you file this issue in the Plex Inc's PMP GitHub, closing issue.

closed 2015-12-17 by NedtheNerd