RasPlex/RasPlex #450: Connecting airplay crashes pi everytime

andebor, 2015-12-18:
While trying to connect my macbook pro(late 2013-Yosemite) to rasplex airplay by using [Airfoil](https://www.rogueamoeba.com/airfoil/), the pi crashes everytime I select it as an output device. It happens instantly. When i try to connect to the the rasplex with OS X built-in airplay support, nothing happens. It won't connect(no error message in OS X) and nothing appears in the rasplex logs. Unfortunately I don't own any other apple devices to perform more tests from.

This is the first time I've tried using the airplay functionality, so I'm not sure when the bug was introduced.

Let me know if there are any more tests that should be run to elaborate.

Serial: 00000000268a9e42
crashid: 16786
Raspberry Pi 2 B
Transcend SDHC class 10 16GB
ethernet connection
Rasplex v1.0.1
Log: http://pastebin.com/WeeDcHXb
Installed with dd on OS X, then upgraded with built-in updater

Power supply: iPad charger 5V, 2A
PlexServer Version 0.9.12.19 running on Debian Wheezy

g1t-dlanor, 2015-12-19:
Prompted by your report I have investigated the airplay behavior of several RasPlex versions today and come to the conclusion that the broken airplay functionality applies to the releases made after RasPlex v0.7.1, which is the last version with which I was able to play music sent from the iTunes app (using OSX Yosemite under VMware). The oldest version which fails to do so for me is the unpublished RasPlex v0.7.2 RC, which was a team-internal version preceding the release of RasPlex v1.0.1, so the critical code change breaking airplay for RasPlex was made somewhere between v0.7.1 and that v0.7.2 release candidate.
Knowing this it will hopefully be possible to fix the problem shortly for a new release.

Additional notes for the programmer:
In my own tests it was not the connetion attempt as such which crashed RasPlex, but the playing of the first MP3 after making the connection. But if the connection is made while already playing something, these events would coincide.
Also, in my tests it was not the OS that crashed (as no full reboot was made). It was the PHT application that crashed, and it was then automatically relaunched by the still running OS, starting with a crash error message. So we're not looking at a full system crash, but a problem internal to the application.

NedtheNerd, 2016-01-11:
See issue #390