RasPlex/RasPlex #452: No thumbnail

Sytten, 2015-12-20:
Hi!

I just installed RasPlex on my PI B+ and I'm experiencing a strange bug.
None of my movies has a thumbnail (with or without pre-caching)
[image]

I can play the movies just fine and the thumbnails appear on the mobile app/webapp.
If someone has an idea on how to fix that I would appreciate.
Thanks!

NedtheNerd, 2015-12-28:
Possible SD card corruption, have you tried reflashing?

Sytten, 2016-01-01:
Yes I just did it and I still have the problem... I will check the debug logs to see if I can find something...

Sytten, 2016-01-01:
I found a lot of warnings concerning, I think, thumbnails like this one:
17:08:12 T:1969668096 WARNING: Skin has invalid include: ListSelectedSquareItemThumb

And then a bunch of errors like that:
-DEBUG: CurlFile::Open(0x601fec88) https://192-168-2-202.b322c72f36074a4f8444e6a11b0b2390.plex.direct:32400/photo/:/transcode?height=320&url=http%3a%2f%2f127.0.0.1%3a32400%2flibrary%2fmetadata%2f11%2fthumb%2f145052$
-ERROR: CPlexTextureCacheJob::CacheTexture unable to open input file plexserver://46e372f3c8cde72c84b6e0ef36c417b176a40a51/photo/:/transcode?height=320&url=http%3a%2f%2f127.0.0.1%3a32400%2flibrary%2fmetadata%2f11%2$

Now if I try to curl the file, I get a 401 error...

NedtheNerd, 2016-01-01:
Have you checked permissions on the folders that contain your libraries, user Plex needs to be able to read them.

Sytten, 2016-01-02:
When I do getent passwd I dont get the user Plex...
On the folder the permissions are:
drwxr-xr-x    2 root     root

NedtheNerd, 2016-01-02:
Then that's your problem, change the owner to plex and group to plex.

Sytten, 2016-01-03:
I would like to, but my passwd file says that this user doesn't exist. And I mean, all the other functionalities work and their permission is set to root owner and group, so my guess is that this is not a problem. There is probably a problem on the server-side not supplying the thumbnails properly, I'm running it on a freebsd box in a jail. I will check the logs on my server to see what happens when the PI makes the request.

Sytten, 2016-01-03:
The server seems to respond:
Jan 03, 2016 07:12:46 [0x80ad29000] DEBUG - Request: [192.168.2.101:52122] GET /photo/:/transcode?height=2048&url=http%3a%2f%2f127.0.0.1%3a32400%2flibrary%2fmetadata%2f9%2fart%2f1450529027&width=2048&X-Plex-Token=xxxxxxxxxxxxxxxxxxxx (5 $
Jan 03, 2016 07:12:46 [0x80ad29000] DEBUG - Auth: We found auth token (xxxxxxxxxxxxxxxxxxxx), enabling token-based authentication.
Jan 03, 2016 07:12:46 [0x80ad29000] DEBUG - Auth: Came in with a super-token, authorization succeeded.
Jan 03, 2016 07:12:46 [0x80ad29000] DEBUG - Photo transcoder: Request for url [http://127.0.0.1:32400/library/metadata/9/art/1450529027] (is local: 0 upscaled: 0)
Jan 03, 2016 07:12:46 [0x80ad29000] DEBUG - HTTP requesting GET http://127.0.0.1:32400/library/metadata/9/art/1450529027
Jan 03, 2016 07:12:46 [0x80a532c00] DEBUG - Request: [192.168.2.202:52251] GET /library/metadata/9/art/1450529027 (5 live)
Jan 03, 2016 07:12:46 [0x809c0d800] DEBUG - Completed: [192.168.2.202:52251] GET /library/metadata/9/art/1450529027 (5 live) 0ms 231 bytes 401
Jan 03, 2016 07:12:46 [0x809c0d800] DEBUG - Completed: [192.168.2.101:52122] GET /photo/:/transcode?height=2048&url=http%3a%2f%2f127.0.0.1%3a32400%2flibrary%2fmetadata%2f9%2fart%2f1450529027&width=2048 (5 live) TLS 3ms 296 bytes 404 (pip$

NedtheNerd, 2016-01-11:
This is as ever related issue, not a Rasplex issue, closing.

closed 2016-01-11 by NedtheNerd

Sytten, 2016-01-11:
Not sure of that, as the server is able to provide correctly the thumbnails to all the other devices except rasplex

NedtheNerd, 2016-01-11:
If this was a Rasplex issue then it would affect every one, not just one or two individuals.  Sorry.

mgbowman, 2016-02-06:
@Sytten did you ever figure this out? I'm having the exact same problem.

mgbowman, 2016-02-06:
@NedtheNerd are you sure? 

After digging a little bit deeper, the debug logs show RasPlex trying to access the thumbnail URLs using an X-Plex-Token=SECRETSTUFF. Is this being stripped out for security reasons? If not, I would think this is the cause of the issue.

According to https://support.plex.tv/hc/en-us/articles/201638786-Plex-Media-Server-URL-Commands, if PMS is part of a Plex Home (which mine is), the X-Plex-Token is required for all remote access.

Will keep digging...

NedtheNerd, 2016-02-06:
Rasplex uses the token, it only writes the token to the log if you enable it in PMS, otherwise it is written solely to the log as SECRETSUFF, this is done for security reasons as people publish logs in the forums when asking for help, it the real token was written to the log others could access the users PMS using the token.

NedtheNerd, 2016-02-06:
I have searched the forums, there are quite a lot of posts about this problem across most of the Plex clients, including 'official' clients, not one of them seems to have been answered with a definitive answer, especially no help from Plex employees or 'ninjas', as I have said before, this is not a Rasplex issue.

NedtheNerd, 2016-02-06:
See this topic https://forums.plex.tv/discussion/136359/thumnails-posters-do-not-update-while-using-pin-bug#latest

mgbowman, 2016-02-06:
I came across that thread and many others but I'm still coming up empty.

Following your advice, I enabled "Allow Plex Media Server tokens in logs" in PMS, restarted both PMS and RasPlex, and the logs still show X-Plex-Token=SECRETSTUFF.

Obviously if I copy and paste the URLs from the logs in a browser, I get 401 Unauthorized because the token is wrong.

I believe you're right that this is not a problem with RasPlex. My guess is this is a legitimate bug in the PMS whenever it's part of a Plex Home.

NedtheNerd, 2016-02-06:
Did you restart PMS and Rasplex after enabling?

NedtheNerd, 2016-02-06:
Sorry just seen you did.

Sytten, 2016-02-06:
No I still have my error, I will check the link when I have some time...

mgbowman, 2016-02-06:
@Sytten are you part of a Plex Home? This would at least help us figure out if this is only an issue with Plex Home accounts.

Sytten, 2016-02-06:
Plex home? I have my own PMS if thats what you mean. Its running on a freebsd server in a jail. Since I dont have any problem with my other devices, I think its a problem with the rasplex query or something...

mgbowman, 2016-02-07:
@Sytten what I was asking was whether the Plex account you're using when you sign in on your PMS and RasPlex is a member of a "Plex Home" account? 

What is Plex Home: https://support.plex.tv/hc/en-us/articles/203815766-What-is-Plex-Home-

I also tried signing-in to both PMS and RasPlex with an account that's not part of a Plex Home and everything works as expected. 

I also experience this same issue (thumbnails not loading) on the official iOS Plex client on my iPad. This is not a RasPlex specific issue.

Sytten, 2016-02-07:
No I'm using only the free version, so without a Plex Home. 
Maybe not a Rasplex issued then... We should contact the dev of plex to see what they cant do...

Sytten, 2016-02-13:
OK thats strange, I tried the home theater app on my laptop and its doing the same thing... confirms that its a server side problem...

mgbowman, 2016-02-17:
@Sytten See http://forums.plex.tv/discussion/comment/1036178/#Comment_1036178

Sytten, 2016-02-18:
I dont have the permission to see this comment...

mgbowman, 2016-02-18:
TL;DR: Add a 127.0.0.1 alias to your jail. This solved it for me.

Sytten, 2016-02-22:
Haaaa, will it work if I have multiple jail with the same loopback address?

Montago, 2020-06-09:
> TL;DR: Add a 127.0.0.1 alias to your jail. This solved it for me.

what ?