RasPlex/RasPlex #453: SSH insecure by default

cyclotron3k, 2015-12-27:
SSH is enabled by default, the default login details are widely published. People are exposing their RasPlex installs to the web.

Even if they aren't putting them in a DMZ, or otherwise exposing them to the web, it's still horribly insecure by default.

Bizarrely, you're prevented from changing the default password because `passwd` is missing.

Possible solutions:
a) Make SSH off by default, force a password to be set the first time SSH is enabled
b) Generate a random password and show it in settings.

g1t-dlanor, 2015-12-27:
I agree that part of your suggestions would be an improvement, though I don't feel that the security issue is as bad as you describe it. RasPlex as a whole is not necessarily exposed to the "web" but only to the LAN it is connected to. Whether or not that LAN allows any SSH access from the Internet is an entirely different matter. Using normal routers there will be a NAT interface between the Internet and all devices on the LAN, and incoming SSH calls from Internet will not automatically be routed to any of those devices without first defining a 'port forward' rule in the router settings.

It's only for cases where people connect the RPi directly to an incoming ISP line, without any intervening LAN router, that the situation you worry about would arise. Though I suppose it could also be an issue where a larger group share a common LAN (like some college dorms etc).

That said, and like I opened with in this post, I do agree that your suggestions are valid and should be considered. It should at any rate be possible to set a password and enforce its use.

But having SSH disabled entirely at first boot is not a good idea in my opinion, as some people may depend on it for initial custom configuration. (Like if they get a launch without visible video, due to compatibility issues with special monitors or projectors.) Instead the initialization wizard could have an additional screen warning users that they should change the SSH password to make it fully secure.

NedtheNerd, 2015-12-27:
Rasplex is OpenELEC OS with Plex HT, so here's the extract from the OpenELEC support pages:

"At the moment it's not possible to change the root password as it's held in a read-only filesystem. However, for the really security conscious advanced user, you can change the password if you build OpenELEC from source. Also you can consider logging in with ssh keys and disabling password logins."

which can be found here: http://wiki.openelec.tv/index.php?title=OpenELEC_FAQ#OpenELEC_and_Kodi

If and when OpenELEC change this it will find its way into Rasplex.

Closing issue.

closed 2015-12-27 by NedtheNerd

dalehamel, 2015-12-27:
As the root file system is read only, there's very little damage to be done.

What rasplex deployment exposes it to a WAN?

The assumption is that rasplex operates from a trusted network.