RasPlex/RasPlex #454: RasPlex 1.0.1 - Crash while playing music and SSH into the RPi2

trumpy81, 2015-12-28:
Serial Number = Where do I find that?
Crash ID = 3088
Size of Pi = Raspberry Pi 2 Model B
SD card = SanDisk Ultra Class 10 8GB
Connection = WiFi
RasPlex = 1.0.1
Logs = don't know how to grab the logs?
Installed = Get-RasPlex 1.0.1
Description = I was playing a music track from my PC PMS install and I SSH'ed into the RPi2 with PuTTY in order to download the RemotePi configuration tool. The tool downloaded, I chmod'ed the file and then ran ./mslconfigtool -l to check the settings, the settings displayed correctly in PuTTY and then RasPlex crashed.

It returned to the RasPlex screen not the OpenELEC screen and rebooted into RasPlex. The crash report dialogue then displayed with the crash ID of 3088.

If there is anymore information needed, just let me know. Thanks.

NedtheNerd, 2016-03-08:
@trumpy81 

Is this still happening on 1.0.3?

trumpy81, 2016-03-08:
I haven't upgraded to 1.0.3 yet and the rasplex.com site is down, not reachable for some reason.

1.0.2 has been working flawlessly for me no issues apart from a single crashes while navigating through my TV Show library, which I suspect was network related.

When I am able to install 1.0.3 I will test it and see if I get the same issue or not.

trumpy81, 2016-03-09:
I upgraded to 1.0.3, finally, and no sign of this issue at all so far. Everything is working well.

This can be closed. :)

NedtheNerd, 2016-03-09:
Closing issue.

closed 2016-03-09 by NedtheNerd