RasPlex/RasPlex #455: [Feature Request] Support for pimoroni pDAC board

kula22, 2016-01-07:
Please can we see this hardware supported in the core image?? Make the Pi0 really useful as a tiny Rasplex client.

NedtheNerd, 2016-01-11:
Have you actually tried this device attached to a RPi0?

kula22, 2016-01-11:
Yes, I've tried it with a pHAT DAC and latest Rasplex. I've also tried running the shell script on the pimoroni site and can't get that to run - I'm sure more due to my lack of rasplex knowledge.

I've tested the pHAT on a Pi2 and the Pi0 using the latest Rasbian jessie and the script works and I get sound out.

NedtheNerd, 2016-01-11:
If someone would like to make a donation to the project to purchase one of these please see details on rasplex..com

NedtheNerd, 2016-03-09:
@kula22 

You don't need to run any scripts, just ensure you have dtoverlay=hifiberry-dac in your config.txt file and reboot, then in Preferences->System->Audio output->Audio output device select 'ALSA: Default (snd_pi_hifiberry_dac Analog'

Closing issue

closed 2016-03-09 by NedtheNerd

kula22, 2016-03-25:
At last got around to testing this and it works a treat!, Excellent, thanks Ned.