RasPlex/RasPlex #456: Crashes while trying to AirPlay

Nosskirneh, 2016-01-11:
Serial number: 00000000fb17fb00
Crash ID: 7942, 7943, 7944, 7947, 7950, 7952, 7956
Size (memory): 1 GB
Size of SD card: 8 GB (microSDHC UHS-I)
Internet connection: Ethernet
Version of RasPlex: 1.0.1
Installed through: GetRasplex (OS X)
Description: When trying to airplay audio to RasPlex, the raspberry pi just dies and reboots.

Steps to reproduce:
1. Install AirParrot 2 (7 days trial).
2. Choose “Audio Only (System Audio)”
3. Choose device “RasPlex”
4. ???
5. Non-Profit

One thing to note is that other iDevices cannot find RasPlex to AirPlay to. Only AirParrot 2 did.

Version of Plex Server: 0.9.12.13
Type of Plex Server: Arch Linux (local server) running on a HP 8100 SFF (model with Intel i5 and 6 GB RAM)
Power supply type: Not an issue.

Video proof:
https://www.dropbox.com/s/1rsxy34kksedsup/IMG_0170.MOV?dl=0

Log file only says this:
http://pastebin.com/1u7Rd8mV

NedtheNerd, 2016-01-11:
See issue #390

Nosskirneh, 2016-01-11:
Ah, thanks for the quick reply. Sorry for duplicate, I tried to google but to no avail. Thanks!

closed 2016-01-11 by Nosskirneh