RasPlex/RasPlex #457: [Feature Request] Change TV Recently Added to season instead of episode

benjaminwolf, 2016-01-11:
Is it possible to change the Recently Added section to display a single season or show icon for recently added media? When you add a whole season/series to PMS, it pushes all other recently added media off of the section.

edjonesy, 2016-04-14:
This would be an excellent improvement.

sguilly123, 2016-06-08:
I also would like to see this implemented.

Electronickss, 2016-09-21:
Have there been any improvements on this?

bkury, 2016-10-04:
done ...

https://github.com/bkury/OpenPHT/commit/7aadfd804599e35b4625f0dc637d7c34333415f5
https://github.com/bkury/OpenPHT/commit/89d3a32571568d98f451d6cb9e8a180b9ff96107

closed 2016-10-04 by bkury

j3g, 2016-10-09:
excellent news!
what build can we find the fix?

bkury, 2016-10-10:
There are no official/stable builds yet but this feature will be in OpenPHT v1.7.0

If you don't mind testing a pre-release, you could also take a look here :
https://github.com/bkury/OpenPHT/releases