RasPlex/RasPlex #458: Installer not working

prudencioj, 2016-01-12:
I'm getting these erros when running the installer in Mac OS:

QSslSocket: cannot resolve TLSv1_1_client_method
QSslSocket: cannot resolve TLSv1_2_client_method
QSslSocket: cannot resolve TLSv1_1_server_method
QSslSocket: cannot resolve TLSv1_2_server_method
QSslSocket: cannot resolve SSL_select_next_proto
QSslSocket: cannot resolve SSL_CTX_set_next_proto_select_cb
QSslSocket: cannot resolve SSL_get0_next_proto_negotiated
Something went wrong with the get request: "Error downloading http://updater.rasplex.com/devices - server replied: Not Found"

NedtheNerd, 2016-01-12:
Working fine on El Capitan and Mavericks.  Could be your OSX setup. Closing issue as there is no issue to fix.  If you want to get help from others as to what the problem may be with your OSX install suggest posting in a topic on the forums which are more likely to be read.

Regards

closed 2016-01-12 by NedtheNerd

dalehamel, 2016-01-12:
This was a legitimate problem with the update server ,it has been fixed.