RasPlex/RasPlex #460: RasPlex @RPI2 Crash when start Playback TV Show

Massaguana, 2016-01-24:
Hello,

here are some Crash´s with RasPlex 1.0.1 runs on my Raspberry Pi 2. Crash id 6045 & 6051.

When i start Playpack for an TV Show RasPlex crash´s and the RPI2 restarts and show´s me the Crash id. At this point it is not possible to use Plex. I hear the navigation sounds but nothing happend. I must restart it my hand... After Restart i must login again and Setup RasPlex new. Skin, and all other Settings are away.

For me it looks like the PHT issue when entering TV Shows and Plex crash´s and restarts... The Same issue with my old RPI, but without an crash id

regards
Massaguana

Massaguana, 2016-02-12:
Here are some more Information´s:

RaspberryPi2 Serial: 00000000fddb4e2a
Crash ID: 6045 & 6051
Raspberry: Pi 2 512MB
SD Card: SDHC Class 10
Connection: Ethernet
Version: RasPlex 1.0.2
How did you install rasplex? win32 disk imager
Power Supply 2A

Logfile are not possible... after Restart there are only the new one in there...

Massaguana, 2016-02-13:
Here are fresh debug logs with this issue
[plexhometheater.old.log.zip](https://github.com/RasPlex/RasPlex/files/129159/plexhometheater.old.log.zip)

Kwiboo, 2016-03-08:
From the log file it looks like the audio player trying to play a video, very weird.
Did you play music before it crashes? Also try to disable theme music in Preferences and see if the crashes goes away.

Massaguana, 2016-03-08:
Rasplex Play's the TV-Show Theme Musik bevor i Start Playing an Episode. Would fit to that if I wait longer, it usually works. In Moment the Theme Musik are disabled, and today the are no Crashs... The other Issue #463 are still there...

Kwiboo, 2016-03-08:
I have noticed some crashes on shutdown after a theme music has been played but not been able to track down and fix the root cause. With the next release (1.6) having the same audio engine as Kodi Jarvis I am hoping similar problems should be resolved or at least more easily resolved.

NedtheNerd, 2016-06-08:
Is this still an issue in Release 1.6.0?

Massaguana, 2016-06-08:
don´t know... can´t test if this issue are away with 1.6.0... to many other issues with it

NedtheNerd, 2016-06-08:
Cannot explain why you are having so many issues, this must be related to hardware, SD card, power supply, power supply cable, using wifi instead of Ethernet (recommended), Preferences settings, TV, AV Receiver, etc. Over 50,000 users are happy with it!

NedtheNerd, 2016-06-10:
Issue cannot be replicated, possible cause are hardware, SD card, power supply, power supply cable, working OK for all other users, closing issue.

closed 2016-06-10 by NedtheNerd