RasPlex/RasPlex #461: Rasplex 1.0.1 crash while playing music - Visualization issue

tas05, 2016-01-29:
Serial number of your pi - (000000005408922d) 

Size of Raspberry Pi (512)

Type of SD card (SDHC 32gb class 10)

Internet connection type (Ethernet)

Version of RasPlex (1.0.1)

Log files (http://pastebin.com/xDMxn3vk)

How did you install rasplex? (gui installer)

Best description you can give of the problem

when I select an album or some music to play, i hear about a second of it and then it crashes going back to Rasplex logo. This seems to be the case for anything i play (all mp3s or Flac). I do not have any issues playing films or TV series. Just Music is affected. The issue was related to having set the visualization (in preferences/settings/music) to OpenGL Spectrum. This was causing the crashing. Once i put it to Now playing everything works as expected.

Kwiboo, 2016-01-29:
All built-in visualization except now playing will be removed in 1.0.2, should fix this issue.

NedtheNerd, 2016-01-30:
Release 1.0.2 removed all visualisations except now playing, closing issue

closed 2016-01-30 by NedtheNerd