RasPlex/RasPlex #462: [Feature Request] "Go To" seeking

samshoo, 2016-01-30:
It would be nice to have a "Go To" seeking feature. The feature should:

- be available via the pause menu
- allow the user to enter a specific time with format HH:MM

NedtheNerd, 2016-01-30:
This feature is already present in Rasplex, whilst a video is playing enter a time (in minutes:seconds, i.e. 6921, to go to 69:21) using the numeric digits that you wish to go to (this will be displayed in the upper left screen and press enter, it will jump to the time you entered.

Closing issue

closed 2016-01-30 by NedtheNerd

samshoo, 2016-01-30:
Wow thanks, is that documented anywhere ? Cause it's like a "hidden" feature, not in menus, hard to find/guess and I wonder if there are other like this one.

NedtheNerd, 2016-01-30:
It was mentioned in the release notes some time ago, sounds like I need to put together a simple functions guide!