RasPlex/RasPlex #463: RasPlex @RPI2 Start Playing Paused

Massaguana, 2016-02-09:
Hello, with the last RasPlex update there are an issue when i Starting the Playback the Playing Starts Paused, i can see the menu at the bottom of the screen...

Kwiboo, 2016-03-08:
Do you have any logs from when playback starts paused, I have not been able to reproduce this locally.
Does it happen for all your media or is it same item that gets paused?

Massaguana, 2016-03-08:
Here are an log with this issue, it´s the last Playing TV-Show Time 21:06... sorry for that big log file, but it´s not so easy to provoke this issue. 
[plexhometheater.zip](https://github.com/RasPlex/RasPlex/files/163884/plexhometheater.zip)

Kwiboo, 2016-03-08:
Thanks, after checking your log-file it seems the player is receiving an extra `Plex.PlayWith(VideoDefaultPlayer)` event when playback is starting and that result in the paused playback.

In the default skin there is only one place where the `Plex.PlayWith(VideoDefaultPlayer)` event is bound to, the play button to the left on the preplay screen.
However in the Plex Black Edition skin you are using there is also a `<onfocus>` that is bound to `Plex.PlayWith(VideoDefaultPlayer)` and my guess it that this gets triggered.
Do you have any disable preplay skin settings activated?

Massaguana, 2016-03-09:
Yes, i use the option´s in Plex Black Edition to hide Preplay and Postplay Screen complete (hide all). I don´t like this... with Plex Black i can see all informations and don´t need an spezial information Step.

Kwiboo, 2016-03-09:
I think that option is the cause of both your issues, the skin is implementing the "disable" function in a way that is causing timing issues (opening the window and using onfocus to quickly starting playback or close the window). I will implement a preference option to disable the preplay and postplay window in a more proper way (not opening the window at all) to work around the need for this type of skin hack.
As a workaround I suggest you (or anyone else having the same issue) to re-enable preplay and use the play-button/p-key to start playback instead of ok/select/enter-key once you have selected an item, that should prevent the preplay window from opening and starting playback directly.

NedtheNerd, 2016-03-09:
deadeyeflint will remove the skin options once Rasplex/OpenPHT has the functionality to hide pre/post play screens.

Quote from him "I'll remove this options when kwiboo adds the function to openpht / rasplex.  I can't find a better way to hide the preplayscreen with skinning only."

Massaguana, 2016-03-11:
Confirm, with preplay and postplay activated there arme no Problems... Now waiting for nee Option to disabled this....

NedtheNerd, 2016-03-11:
New option has been added to next release, as this is a skin issue I'm closing the issue.

closed 2016-03-11 by NedtheNerd