RasPlex/RasPlex #465: Cannot use RasPlex Installer on Windows or OSX - http://updater.rasplex.com/devices - server replied: Not Found

MikeRatcliffe, 2016-02-13:
No options for model in dropdown in Windows or OSX.

OSX Installer output:
$ sudo '/Users/<myusername>/Desktop/GetRasplex.app/Contents/MacOS/rasplex-installer'

Password:
Refreshing removable devices list
Getting  QUrl( "http://updater.rasplex.com/devices" ) 
QSslSocket: cannot resolve TLSv1_1_client_method
QSslSocket: cannot resolve TLSv1_2_client_method
QSslSocket: cannot resolve TLSv1_1_server_method
QSslSocket: cannot resolve TLSv1_2_server_method
QSslSocket: cannot resolve SSL_select_next_proto
QSslSocket: cannot resolve SSL_CTX_set_next_proto_select_cb
QSslSocket: cannot resolve SSL_get0_next_proto_negotiated
Something went wrong with the get request: "Error downloading http://updater.rasplex.com/devices - server replied: Not Found"

Rosefox911, 2016-02-14:
I am experiencing the exact same issue on Windows 7.

g1t-dlanor, 2016-02-14:
Unfortunately there's currently a problem with the update server.
This causes the program to fail finding supported hardware models or download images.
You can then click the "Cancel" button, to break the failed contact attempts.
Thereafter you can still use the installer to select a local image file and flash it to the selected device.

So for the time being you'll need to download the image files 'manually' through a web browser.
You can find them at this web page of github: https://github.com/RasPlex/RasPlex/releases

Hopefully the server issue will soon be fixed so that you can again use all installer features.

dalehamel, 2016-02-14:
Thanks for bringing this issue to our attention, the problem has been resolved.

closed 2016-02-14 by dalehamel