RasPlex/RasPlex #471: [Feature Request] Change FF/RW to timeline scrub using BIF files.

benjaminwolf, 2016-03-06:
This issue was brought up a problem with skipping through or FF/RW transcoded streams from a remote server.

https://forums.plex.tv/discussion/202155/awfully-slow-ff-rw-during-transcoding#latest

A solution to the problem would be to use a timeline scrub method of finding your place in the media file used by some other Plex clients such as the Roku Client.

https://support.plex.tv/hc/en-us/articles/202197528-Video-Preview-Thumbnails

This method avoids the pitfalls of the current skip and FF/RW options by pausing the media stream during the scrubbing and restarting the stream from a specific time within the media file instead of having to start and stop the transcode over and over while looking for the desired scene.

NedtheNerd, 2016-03-09:
When investigating this note that not everyone generates BIF files due to storage impact.

cyrosystems, 2016-04-09:
+1 for this feature please, both local and remote playback fast forward and rewind on a timeline.

Kwiboo, 2016-06-09:
Parts of this feature request is included in the 1.6.0 release.

There is now support for adaptive seeking and once adaptive seeking is performed a small preview thumbnail will be shown if PMS has generated video preview thumbnails.

Kwiboo, 2016-06-30:
I consider this feature request implemented with the final piece being configurable seek steps.

closed 2016-06-30 by Kwiboo