RasPlex/RasPlex #473: Decoding artifacts with old XviD and RasPlex

Massaguana, 2016-03-19:
Hello together,

with RasPlex there are some Problems with very old XviD Movies... With all of them there are massiv decoding artifacts. 

Here are an MediaInfo:

``Allgemein
Vollständiger Name                       : /Volumes/TV-Serien/Star Trek Deep Space Nine/Star Trek DS9 - S01E01-02 - Der Abgesandte (DVD-Rip XviD GER-AC3 ENG-MP3) by mulder77.avi
Format                                   : AVI
Format/Info                              : Audio Video Interleave
Dateigröße                               : 977 MiB
Dauer                                    : 1h 27min
Modus der Gesamtbitrate                  : variabel
Gesamte Bitrate                          : 1 568 Kbps
Kodierendes Programm                     : Nandub v1.0rc2
verwendete Encoder-Bibliothek            : Nandub build 1852/release

Video
ID                                       : 0
Format                                   : MPEG-4 Visual
Format-Profil                            : Advanced Simple@L5
Format-Einstellungen für BVOP            : 2
Format-Einstellungen für Qpel            : Ja
Format-Einstellungen für GMC             : 3 warppoints
Format-Einstellungen für Matrix          : Default (H.263)
Muxing-Modus                             : Packed bitstream
Codec-ID                                 : XVID
Codec-ID/Hinweis                         : XviD
Dauer                                    : 1h 27min
Bitrate                                  : 980 Kbps
Breite                                   : 576 Pixel
Höhe                                     : 416 Pixel
Bildseitenverhältnis                     : 1,385
Bildwiederholungsrate                    : 25,000 FPS
ColorSpace                               : YUV
ChromaSubsampling/String                 : 4:2:0
BitDepth/String                          : 8 bits
Scantyp                                  : progressiv
Bits/(Pixel*Frame)                       : 0.164
Stream-Größe                             : 610 MiB (62%)
verwendete Encoder-Bibliothek            : XviD 1.0.3 (UTC 2004-12-20)

Audio #1
ID                                       : 1
Format                                   : AC-3
Format/Info                              : Audio Coding 3
Format_Settings_ModeExtension            : CM (complete main)
Format-Einstellungen für Endianess       : Big
Codec-ID                                 : 2000
Dauer                                    : 1h 27min
Bitraten-Modus                           : konstant
Bitrate                                  : 448 Kbps
Kanäle                                   : 6 Kanäle
Kanal-Positionen                         : Front: L C R, Side: L R, LFE
Samplingrate                             : 48,0 KHz
Bildwiederholungsrate                    : 31,250 FPS (1536 spf)
Stream-Größe                             : 279 MiB (29%)
Ausrichtung                              : Aufteilung über Interleaves
Interleave, Dauer                        : 40 ms (1,00 Video-Frame)
Interleave, Vorlaufsdauer                : 500 ms

Audio #2
ID                                       : 2
Format                                   : MPEG Audio
Format-Version                           : Version 1
Format-Profil                            : Layer 3
Format_Settings_Mode                     : Dual mono
Codec-ID                                 : 55
Codec-ID/Hinweis                         : MP3
Dauer                                    : 1h 27min
Bitraten-Modus                           : variabel
Bitrate                                  : 128 Kbps
Kanäle                                   : 2 Kanäle
Samplingrate                             : 48,0 KHz
Stream-Größe                             : 76,2 MiB (8%)
Ausrichtung                              : Ausgerichtet an Interleaves
Interleave, Dauer                        : 24 ms (0,60 Video-Frame)
Interleave, Vorlaufsdauer                : 496 ms
verwendete Encoder-Bibliothek            : LAME3.96r
Kodierungseinstellungen                  : -m d -V 4 -q 3 -lowpass 17.5 --abr 128``

Regards
Massaguana

Massaguana, 2016-03-20:
With PMP 1.0.6 there are the same issue... Maybe an Raspberry problem?

NedtheNerd, 2016-03-20:
There is a known problem on the RPi with old DIX3 files, even using Raspbian, have you tried using the PMS optimise to convert the file?

Closing issue.

closed 2016-03-20 by NedtheNerd

Massaguana, 2016-03-20:
Currently I encode all new from DVD... a lot of work.