RasPlex/RasPlex #474: Transcode DTS to AC3

ZyberSE, 2016-03-21:
RasPlex 1.0.3 on a RPi2 can't re-encode DTS to AC3 and send it over HDMI. It sounds like it just takes the rear channels and output them as a simple stereo stream.

My SONOS system confirms that the signal is a normal stereo stream when playing DTS, but when playing AC3 it's Dolby Digital 5.1 as expected.

NedtheNerd, 2016-06-06:
Can you try latest 1.6.0 release and advise.

ZyberSE, 2016-06-07:
I just re-installed 1.6 and I still do not get DTS -> AC3 transcoding to work.
Is there any particular settings I need to adjust?

ZyberSE, 2016-06-08:
Just tried the same thing on a new RPi 3 with the exact same result :(

Kwiboo, 2016-06-08:
You need to activate "Enable Dolby Digital transcoding" in the audio output preferences.
See http://kodi.wiki/view/Audio_settings for an overview of all options, OpenPHT uses similar options.

ZyberSE, 2016-06-08:
I can't find that option in the GUI?

NedtheNerd, 2016-06-08:
Near the bottom of Preferences-System-Audio output.

ZyberSE, 2016-06-08:
I have...
Enable passthough
Passtrough output device
-- DD (AC3) cap.. rec..
-- DD Plus
-- DTS

NedtheNerd, 2016-06-08:
Set speakers to 2.0, enable Passthrough, check Dolby Digital (AC3) capable receiver, check Enable Dolby Digital transcoding.

ZyberSE, 2016-06-09:
Wohoo, it's working! :)
(but I would never had figured it out without your instructions, the gui could be more self-explanatory)

NedtheNerd, 2016-06-09:
Closing issue

closed 2016-06-09 by NedtheNerd

NedtheNerd, 2016-06-09:
@ZyberSE 

I have updated my Rasplex Audio Guide on the forums.