RasPlex/RasPlex #475: Content Version Selection Doesn't Prompt When Cinema Trailers Are Enabled

magaman, 2016-03-26:
If cinema trailers are enabled there is no prompt when starting a film to select which version to play if multiple versions are available (ie 1080p h264 and a 4K h265 version) it appears to default to highest resolution file. Disabling cinema trailers corrects the issue.

NedtheNerd, 2016-04-01:
According to Plex Inc it should automatically select the highest available that is supported by the client.

magaman, 2016-04-02:
And it is playing the highest available, but if you disable cinema trailers when you select play you get a prompt asking which version. This does not occur with cinema trailers enabled. Also the Plex Media app presents the prompt when trailers are enabled. The reason this can be an issue is in the case I described where I have a 4K copy available for 4K screens, but if I'm viewing on a 1080p screen there is no reason for me to run a transcode of the 4K version which is what occurs without the prompt available.