RasPlex/RasPlex #476: Hyperion in autostart.sh

werkkrew, 2016-03-28:
On my recent install (v1.0.3) it appears that when enabling Hyperion there is a conflicting systemd service and entry in /storage/.config/autostart.sh leading to two conflicting instances of hyperion running.

Removing the entry from autostart.sh in favor of the systemd service fixes any issues caused by this.

Without doing another fresh install I cannot be sure if the autostart.sh entry is there from the time of installation or upon the first time a user enables the hyperion service in the system settings. 

I can however confirm that subsequently disabling/enabling the service in system settings does not re-add the entry to autostart.sh

Kwiboo, 2016-03-31:
A clean RasPlex install should not have a `autostart.sh`-file and the hyperion option in System Settings only enable/disables the systemd service for the bundled hyperiond by modifying `/storage/.cache/services/hyperion.conf`.

Your `autostart.sh`-file must have come from somewhere else, have you tried to install a second hyperiond using the `install_hyperion.sh` script?

NedtheNerd, 2016-04-01:
Closing issue.

closed 2016-04-01 by NedtheNerd

werkkrew, 2016-04-01:
I am not sure why Ned would just close this, I had the autostart.sh file on an absolutely clean install of the newest release.  It was not a relic of an old install and it was not something I manually created.

NedtheNerd, 2016-04-01:
I checked this with a clean install, no autostart.sh file is created. Enabled Hyperion support in System Settings, still no autostart.sh file created.  Therefore this must be created by something that you have done, did you follow this guide https://forums.plex.tv/discussion/180097/hyperion-lightberry-hd-on-rasplex-setup-instructions-default-settings-plus-hints-tips/p1