RasPlex/RasPlex #478: Unable to get code or to log in manually

SafaAlfulaij, 2016-03-31:
I have a Raspberry Pi 2 model. Version of RasPlex: devel-20151128145708-r20356-gdbb11b7
I am using wifi connection.
Whenever I try to sign in to Plex, the loader icon keeps spinning. I've tried both the pin key and the manual method.

PIN key log:
```
00:33:44 T:1970839552   DEBUG: OnKey: return (f00d) pressed, action is Select
00:33:44 T:1970839552   DEBUG: Activating window ID: 10203
00:33:45 T:1970839552   DEBUG: ------ Window Deinit (SettingsCategory.xml) ------
00:33:45 T:1970839552   DEBUG: ExecuteXBMCAction : Translating Skin.Reset(SettingsCategory)
00:33:45 T:1970839552   DEBUG: ExecuteXBMCAction : To Skin.Reset(SettingsCategory)
00:33:45 T:1970839552   DEBUG: ------ Window Init (MyPlexLogin.xml) ------
00:33:45 T:1970839552    INFO: Loading skin file: MyPlexLogin.xml, load type: LOAD_EVERY_TIME
00:33:45 T:1970839552   DEBUG: Using already stored xml root node for /usr/share/XBMC/addons/skin.plex/720p/MyPlexLogin.xml
00:33:45 T:1970839552   DEBUG: CMyPlexManager::BroadcastState state: fetching PIN, error: network
00:33:45 T:1811657808   DEBUG: CPlexFile::BuildHTTPURL translated 'plexserver://myplex/pins.xml' to 'https://plex.tv:443/pins.xml'
00:33:45 T:1811657808   DEBUG: CurlFile::Open(0x6bfbac14) https://plex.tv:443/pins.xml
00:33:45 T:1811657808    INFO: easy_aquire - Created session to https://plex.tv
00:33:45 T:1786491984   DEBUG: Mimetype: application/octet-stream
00:33:45 T:1786491984   DEBUG: CPlexDirectory::GetDirectory plexserver://best:32400/library/arts/?sort=random&X-Plex-Container-Start=0&X-Plex-Container-Size=50
00:33:45 T:1786491984   DEBUG: CPlexServer::BuildURL no active connections for Safa version:  activeConnection: NO owner: you deletion: YES class:
00:33:45 T:1786491984   DEBUG: CCurlFile::Service reading extra failure information
00:33:45 T:1786491984   DEBUG: CPlexDirectory::GetDirectory failed to fetch data from plexserver://best:32400/library/arts/?sort=random&X-Plex-Container-Start=0&X-Plex-Container-Size=50: -1
00:33:45 T:1786491984   ERROR: GetDirectory - Error getting plexserver://best:32400/library/arts/?sort=random&X-Plex-Container-Start=0&X-Plex-Container-Size=50
00:33:45 T:1811657808 WARNING: CCurlFile::FillBuffer [https://plex.tv:443/pins.xml] failed: Peer certificate cannot be authenticated with given CA certificates
00:33:45 T:1811657808   DEBUG: CCurlFile::Service reading extra failure information
00:33:45 T:1811657808   ERROR: CMyPlexManager::GetXml failed to fetch plexserver://myplex/pins.xml : -1
00:33:45 T:1811657808   DEBUG: CMyPlexManager::BroadcastState state: refresh, error: network
00:33:45 T:1811657808   DEBUG: CMyPlexManager::Process after a run our state is 0, will now sleep for 3600 seconds
```

Username/Password method log:
```
00:47:31 T:1811657808   DEBUG: CPlexFile::BuildHTTPURL translated 'plexserver://SECRETUSER:SECRETPASSWORD@myplex/users/sign_in.xml' to 'https://SECRETUSER:SECRETPASSWORD@plex.tv:443/users/sign_in.xml'
00:47:31 T:1811657808   DEBUG: CurlFile::Open(0x6bfbab14) https://plex.tv:443/users/sign_in.xml
00:47:31 T:1811657808    INFO: easy_aquire - Created session to https://plex.tv
00:47:32 T:1811657808 WARNING: CCurlFile::FillBuffer [https://SafaAlFulaij:PASSWORDHIDDEN@plex.tv:443/users/sign_in.xml] failed: Peer certificate cannot be authenticated with given CA certificates
00:47:32 T:1811657808   DEBUG: CCurlFile::Service reading extra failure information
00:47:32 T:1811657808   ERROR: CMyPlexManager::GetXml failed to fetch plexserver://SafaAlFulaij:PASSWORDHIDDEN@myplex/users/sign_in.xml : -1
00:47:32 T:1811657808   DEBUG: CMyPlexManager::BroadcastState state: refresh, error: network
00:47:32 T:1811657808   DEBUG: CMyPlexManager::Process after a run our state is 0, will now sleep for 3600 seconds
00:47:38 T:1786491984   DEBUG: Thread Jobworker 1786491984 terminating (autodelete)
00:47:38 T:1837102160   DEBUG: CPlexGlobalTimer::Process firing callback serverDataLoader - spent -1 in queue
00:47:38 T:1837102160   DEBUG: CPlexGlobalTimer::Process waiting 29979 milliseconds for manualServerManager...
00:47:38 T:1786491984  NOTICE: Thread Jobworker start, auto delete: true
00:47:38 T:1786491984   DEBUG: CPlexGlobalTimer::SetTimeout adding timeout: serverDataLoader at pos 1 [30000]

```
My password contains English letters, numbers and some special characters like "#" and "&" (Password is encoded in the URL), and it is correct, I double-checked it :)

Kwiboo, 2016-03-31:
The logs show that there is some certificate validation error with the request to plex.tv: "Peer certificate cannot be authenticated with given CA certificates"

I suggest you upgrade to RasPlex 1.0.3 and try again, change update channel to Pre-release or download latest image from https://github.com/RasPlex/RasPlex/releases

SafaAlfulaij, 2016-03-31:
Same. No change at all. :(

Kwiboo, 2016-03-31:
Could be a time issue, unless your local time was 00:33 or 00:47 at the time of the log snippets above. How is the network configured, static or DHCP?
A full log would also be helpful.

SafaAlfulaij, 2016-04-01:
You were right! It was a time problem! Issue solved!
Any idea how to get the time from the Internet? :)

NedtheNerd, 2016-04-01:
It should pick it up automatically, suggest enabling wait for network in System Settings.

Closing issue.

closed 2016-04-01 by NedtheNerd