RasPlex/RasPlex #479: Auto advance to next episode in season

NedtheNerd, 2016-04-02:
Currently when an episode is played from the pre-play screen and the episode reaches the end you are returned to the post-play screen (which is actually the same screen as the pre-play screen) and then you have to press right to advance to the next episode.

Request the functionality when returning to the pre/post play screen that the next unwatched episode in the season is presented, with the following conditions:

Only display the next episode if the played episode has completed,

If the episode is partially watched then that episode is displayed,

If the episode watched is the last episode in the season then present the first unwatched episode in the next season, if a subsequent season is available,

If the episode watched is the last episode in the season and no other seasons exist return to the Home screen.

edjonesy, 2016-04-14:
That would all be great, maybe as well have an option to enable or disable in the preferences, automatically playing the next episode after 20 seconds, like Netflix does.

ovizii, 2016-08-02:
Seems to have been implemented? Or is this because I updated my PMC to the latest version? At least here, the episode auto advance works.