RasPlex/RasPlex #480: rasplex not connecting to local server as expected when internet went down.

quinnmj, 2016-04-02:
https://forums.plex.tv/discussion/comment/1159617#Comment_1159617.
Net - I believe all my server settings were correct. rasplex wouldn't connect to my server when the internet went down.  Somewhere along the way, my server became "unclaimed".  Don't know if this was required or not.
I had to go into the "Sign in" screen, then cancel out.  At that point, rasplex connected with the server.  This is ugly as hell - it should just automatically connect to my local server (it was manually specified) if it can't connect to plex.

Kwiboo, 2016-04-02:
Please enable debug logging in Preferences and provide a full log file of a reboot when you experience this issue. Also include your workaround of cancel out from the sign in screen actions to get your server to show up in the log file.

NedtheNerd, 2016-06-08:
Is this issue resolved?

NedtheNerd, 2016-06-10:
No further comments received, closing issue.

closed 2016-06-10 by NedtheNerd